slightech/MYNT-EYE-S-SDK #3: help, an error occur when run 'make ros'

ZhaoQiang2015, 2018-06-04:
the output error follows this:

make[1]: Entering directory '/home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/build'
-- Using CATKIN_DEVEL_PREFIX: /home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /home/zq/anaconda3/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /home/zq/anaconda3/bin/nosetests
-- catkin 0.7.6
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - mynt_eye_ros_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'mynt_eye_ros_wrapper'
-- ==> add_subdirectory(mynt_eye_ros_wrapper)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found OpenCV: 3.2.0
-- mynt_eye_ros_wrapper: 1 messages, 0 services
-- Found glog: 0.3.5
-- Found mynteye: 2.0.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/build
make[1]: Leaving directory '/home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/build'
####
#### Running command: "make -j8 -l8" in "/home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/build"
####
make[1]: Entering directory '/home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/build'
make[2]: Entering directory '/home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/build'
make[3]: Entering directory '/home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/build'
make[3]: Leaving directory '/home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/build'
make[3]: Entering directory '/home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/build'
[  9%] Building CXX object mynt_eye_ros_wrapper/CMakeFiles/mynteye_wrapper.dir/src/wrapper_nodelet.cc.o
/home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/src/mynt_eye_ros_wrapper/src/wrapper_nodelet.cc:26:39: fatal error: mynt_eye_ros_wrapper/Temp.h: No such file or directory
compilation terminated.
mynt_eye_ros_wrapper/CMakeFiles/mynteye_wrapper.dir/build.make:62: recipe for target 'mynt_eye_ros_wrapper/CMakeFiles/mynteye_wrapper.dir/src/wrapper_nodelet.cc.o' failed
make[3]: *** [mynt_eye_ros_wrapper/CMakeFiles/mynteye_wrapper.dir/src/wrapper_nodelet.cc.o] Error 1
make[3]: Leaving directory '/home/zq/Downloads/SDKs/MYNT-EYE-SDK-2/wrappers/ros/build'
CMakeFiles/Makefile2:435: recipe for target 'mynt_eye_ros_wrapper/CMakeFiles/mynteye_wrapper.dir/all' failed

I would very thanks for your help!

joinAero, 2018-06-04:
You could change the jobs of `cakin_make` and try again.

* Edit `Makefile`, change `catkin_make` to `catkin_make -j2`, then `make ros` again.

```
ros: install
	@$(call echo,Make $@)
ifeq ($(HOST_OS),Win)
	$(error "Can't make ros on win")
else
	@cd ./wrappers/ros && catkin_make -j2
endif
```

`Temp.h` is a ROS message. Normally, it should be generated by the `catkin_make`.

ZhaoQiang2015, 2018-06-05:
Thank you very much! @joinAero 
I modify the Makefile follow your advice, and it's OK now!

but some confused with why it could affect the make result with different threading -j. Have you ever met with this error?

closed 2018-06-05 by yangruixuan