slightech/MYNT-EYE-S-SDK #5: error: firmware upgrade to latest version

rising-turtle, 2018-06-19:
Hi all:
I compiled the SDK-2 and made samples. But when running the executable program in the sample folder like "camera_d" or "ctrl_manual_exposure", it stopped and complained the following error

> F0618 22:50:58.153812  5142 device.cc:544] Read device infos failed. Please upgrade your firmware to the latest version.

Then I updated the firmware in windows 7 according to the instructions in this page https://slightech.github.io/MYNT-EYE-SDK/upgrade_firmware.html . I used the latest image (mynteye_0106_1803201637.img) listed in this page https://github.com/slightech/MYNT-EYE-SDK/tree/master/firmware 

However, after the update, the executable program "camera_d" still cannot run, complaining the same error. 

How can I solve this issue?  Any suggestion is greatly appreciated. 
Thanks.

rising-turtle, 2018-06-19:
Besides, it also output warnings, like 

> W0618 23:04:44.311489  7398 channels.cc:1014] Get PuControlInfo of Option::GAIN failed
W0618 23:04:44.311514  7398 uvc-v4l2.cc:207] pu_control_range failed error 22, Invalid argument
W0618 23:04:44.311523  7398 channels.cc:1014] Get PuControlInfo of Option::CONTRAST failed
W0618 23:04:44.311543  7398 uvc-v4l2.cc:236] xu_control_query failed error 2, No such file or directory
W0618 23:04:44.311555  7398 uvc-v4l2.cc:515] xu_control_range query failed

yangruixuan, 2018-07-23:
Please use SDK 2.x with Firmware 2.x

closed 2018-07-23 by yangruixuan