slightech/MYNT-EYE-S-SDK #8: Image packet header must be 0x3B, but 0x00 now

LSXiang, 2018-07-23:
Hi,
I applied the earliest hardware（Purchased on 2017/08） to run SDK2, but the camera could not capture the image, the log is as follows：

$ ./samples/_output/bin/api/camera_a
W0723 21:21:11.942250 15214 streams.cc:53] Image packet header must be 0x3B, but 0x00 now
W0723 21:21:11.942327 15214 streams.cc:163] Image packet is unaccepted, frame dropped
...
...
W0723 21:21:11.942250 15214 streams.cc:53] Image packet header must be 0x3B, but 0x00 now
W0723 21:21:11.942327 15214 streams.cc:163] Image packet is unaccepted, frame dropped
F0723 21:21:11.975056 15205 streams.cc:179] Timeout waiting for key frames. Please use USB 3.0, and not in virtual machine.
*** Check failure stack trace: ***
    @     0x7f8ea1ea28ad  google::LogMessage::Fail()
    @     0x7f8ea1ea4f78  google::LogMessage::SendToLog()
    @     0x7f8ea1ea231b  google::LogMessage::Flush()
    @     0x7f8ea1ea5e4e  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f8ea1e6e80b  mynteye::Streams::WaitForStreams()
    @     0x7f8ea1e7a97d  mynteye::Device::WaitForStreams()
    @     0x7f8ea1e8ae69  mynteye::Synthetic::WaitForStreams()
    @           0x402a5c  main
    @     0x7f8ea0064830  __libc_start_main
    @           0x404dc9  _start
    @              (nil)  (unknown)
Aborted (core dumped)

How can I solve this issue? Any suggestion is greatly appreciated.
Thanks.

LSXiang, 2018-07-23:
I am sure i use USB 3.0, and not in virtual machine. The SDK1.x can run in my computer!!!

yangruixuan, 2018-07-23:
Have you upgraded the firmware to 2.x?

LSXiang, 2018-07-23:
Yes, i have. i follow the _http://doc.myntai.com/resource/sdk/mynt-eye-sdk-guide-2.0.1-rc0-html-zh-Hans/mynt-eye-sdk-guide-2.0.1-rc0-html-zh-Hans/src/firmware/contents.html_ upgraded the firmware.

melodybinbin, 2018-08-02:
Hi, I meet a similar issue.
When I run: roslaunch mynt_eye_ros_wrapper display.launch
It shows as follows in the terminal. 
(Ubuntu 16.04、SDK2.0)

vio@vio:~$ roslaunch mynt_eye_ros_wrapper display.launch 
… logging to /home/vio/.ros/log/d3685832-955a-11e8-9ebb-28c63f8d78e0/roslaunch-vio-20193.log 
Checking log directory for disk usage. This may take awhile. 
Press Ctrl-C to interrupt 
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://vio:43401/

SUMMARY
PARAMETERS

/mynteye/mynteye_wrapper_node/base_frame_id: mynteye_link
/mynteye/mynteye_wrapper_node/brightness: -1
/mynteye/mynteye_wrapper_node/contrast: -1
/mynteye/mynteye_wrapper_node/depth_frame_id: mynteye_depth_frame
/mynteye/mynteye_wrapper_node/depth_topic: depth/image_raw
/mynteye/mynteye_wrapper_node/desired_brightness: -1
/mynteye/mynteye_wrapper_node/disparity_frame_id: mynteye_disparity…
/mynteye/mynteye_wrapper_node/disparity_norm_frame_id: mynteye_disparity…
/mynteye/mynteye_wrapper_node/disparity_norm_topic: disparity/image_norm
/mynteye/mynteye_wrapper_node/disparity_topic: disparity/image_raw
/mynteye/mynteye_wrapper_node/enable_depth: True
/mynteye/mynteye_wrapper_node/enable_disparity: False
/mynteye/mynteye_wrapper_node/enable_disparity_norm: False
/mynteye/mynteye_wrapper_node/enable_left_rect: False
/mynteye/mynteye_wrapper_node/enable_points: False
/mynteye/mynteye_wrapper_node/enable_right_rect: False
/mynteye/mynteye_wrapper_node/exposure_mode: -1
/mynteye/mynteye_wrapper_node/frame_rate: -1
/mynteye/mynteye_wrapper_node/gain: -1
/mynteye/mynteye_wrapper_node/gravity: 9.8
/mynteye/mynteye_wrapper_node/hdr_mode: -1
/mynteye/mynteye_wrapper_node/imu_frame_id: mynteye_imu_frame
/mynteye/mynteye_wrapper_node/imu_frequency: -1
/mynteye/mynteye_wrapper_node/imu_topic: imu/data_raw
/mynteye/mynteye_wrapper_node/ir_control: 80
/mynteye/mynteye_wrapper_node/left_frame_id: mynteye_left_frame
/mynteye/mynteye_wrapper_node/left_rect_frame_id: mynteye_left_rect…
/mynteye/mynteye_wrapper_node/left_rect_topic: left/image_rect
/mynteye/mynteye_wrapper_node/left_topic: left/image_raw
/mynteye/mynteye_wrapper_node/max_exposure_time: -1
/mynteye/mynteye_wrapper_node/max_gain: -1
/mynteye/mynteye_wrapper_node/points_frame_id: mynteye_points_frame
/mynteye/mynteye_wrapper_node/points_topic: points/data_raw
/mynteye/mynteye_wrapper_node/right_frame_id: mynteye_right_frame
/mynteye/mynteye_wrapper_node/right_rect_frame_id: mynteye_right_rec…
/mynteye/mynteye_wrapper_node/right_rect_topic: right/image_rect
/mynteye/mynteye_wrapper_node/right_topic: right/image_raw
/mynteye/mynteye_wrapper_node/temp_frame_id: mynteye_temp_frame
/mynteye/mynteye_wrapper_node/temp_topic: temp/data_raw
/rosdistro: kinetic
/rosversion: 1.12.13
NODES 
/mynteye/ 
mynteye_wrapper_node (mynt_eye_ros_wrapper/mynteye_wrapper_node) 
/ 
rviz (rviz/rviz)

auto-starting new master 
process[master]: started with pid [20203] 
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d3685832-955a-11e8-9ebb-28c63f8d78e0 
process[rosout-1]: started with pid [20216] 
started core service [/rosout] 
[... 29 lines cut ...]
[ INFO] [1533107797.134120129]: Advertized on topic disparity/image_raw 
[ INFO] [1533107797.167372788]: Advertized on topic disparity/image_norm 
[ INFO] [1533107797.201338946]: Advertized on topic depth/image_raw 
[ INFO] [1533107797.202258265]: Advertized on topic points/data_raw 
[ INFO] [1533107797.203001176]: Advertized on topic imu/data_raw 
[ INFO] [1533107797.203795383]: Advertized on topic temp/data_raw 
[ INFO] [1533107797.206701904]: Enable stream data of Stream::DEPTH 
[ INFO] [1533107797.208304924]: Advertized service get_info 
W0801 15:16:37.208601 20233 device.cc:266] Motion extrinsics from Stream::LEFT not found 
[ERROR] [1533107797.208873806]: Ignoring transform for child_frame_id "mynteye_right_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.500000 0.000000 0.000000 0.000000) 
[ERROR] [1533107797.208978402]: Ignoring transform for child_frame_id "mynteye_right_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.500000 0.000000 0.000000 0.000000) 
[ERROR] [1533107797.209078199]: Ignoring transform for child_frame_id "mynteye_right_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.500000 0.000000 0.000000 0.000000) 
[ERROR] [1533107797.209154401]: Ignoring transform for child_frame_id "mynteye_right_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.500000 0.000000 0.000000 0.000000) 
[ERROR] [1533107797.209223205]: Ignoring transform for child_frame_id "mynteye_right_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.500000 0.000000 0.000000 0.000000) 
[ERROR] [1533107797.209292365]: Ignoring transform for child_frame_id "mynteye_right_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.500000 0.000000 0.000000 0.000000) 
[ERROR] [1533107797.209363488]: Ignoring transform for child_frame_id "mynteye_right_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.500000 0.000000 0.000000 0.000000) 
[ERROR] [1533107797.209450249]: Ignoring transform for child_frame_id "mynteye_right_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.500000 0.000000 0.000000 0.000000) 
[... cut ...]  
W0801 15:16:37.297461 20713 streams.cc:163] Image packet is unaccepted, frame dropped 
W0801 15:16:37.422971 20713 streams.cc:53] Image packet header must be 0x3B, but 0x00 now 
W0801 15:16:37.423005 20713 streams.cc:163] Image packet is unaccepted, frame dropped 
W0801 15:16:37.423360 20713 streams.cc:53] Image packet header must be 0x3B, but 0x00 now 
W0801 15:16:37.423378 20713 streams.cc:163] Image packet is unaccepted, frame dropped 

What is wrong with it? Thank you very much in advance!

yangruixuan, 2018-08-10:
This error means the SDK received no data from camera. Please make sure you have connect the camera with cable stably. Maybe you can try to use different cable and USB interface.

LSXiang, 2018-08-10:
I make sure the computer have connected stably the camera by cable. I trｉｅｄ to use all USB3.0 interface in 2 computer, but it didn't work. And then, I used the SDK1.8 sample and the camera is running normally (the firmware has no changed, MYNTEYE_S_2.0.0_rc.img).  Is it possible that the new SDK cannot support old hardware?

WwYyFan, 2018-08-21:
@LSXiang I also encountered this problem. Can you solve it now?

LSXiang, 2018-08-21:
@WwYyFan Not yet.  The old hardware that company purchased is working with SDK1.8 now. The new camera(S1030) I bought privately did not have this problem. ╮(╯▽╰)╭

WwYyFan, 2018-08-21:
@LSXiang Running result in sdk1.8，The following is the output, is it wrong?

roslaunch mynteye_ros_wrapper mynt_camera.launch 
... logging to /home/efun/.ros/log/d8b0f22a-a50b-11e8-98a3-4cedfbcaaae4/roslaunch-efun-13012.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:41287/

SUMMARY
========

PARAMETERS
 * /mynteye_ros_wrapper/depth_topic: depth/depth_regis...
 * /mynteye_ros_wrapper/device_name: 0
 * /mynteye_ros_wrapper/enable_cpu: False
 * /mynteye_ros_wrapper/enable_depth: False
 * /mynteye_ros_wrapper/imu_topic: imu
 * /mynteye_ros_wrapper/left_cam_info_topic: left/camera_info
 * /mynteye_ros_wrapper/left_raw_topic: left/image_raw_color
 * /mynteye_ros_wrapper/left_topic: left/image_rect_c...
 * /mynteye_ros_wrapper/right_cam_info_topic: right/camera_info
 * /mynteye_ros_wrapper/right_raw_topic: right/image_raw_c...
 * /mynteye_ros_wrapper/right_topic: right/image_rect_...
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    mynteye_ros_wrapper (mynteye_ros_wrapper/mynteye_wrapper_node)

ROS_MASTER_URI=http://localhost:11311

process[mynteye_ros_wrapper-1]: started with pid [13029]
[ INFO] [1534836435.477643649]: Initializing nodelet with 12 worker threads.
[ INFO] [1534836435.520491805]: Advertized on topic left/image_rect_color
[ INFO] [1534836435.535172535]: Advertized on topic left/image_raw_color
[ INFO] [1534836435.535535444]: Advertized on topic left/camera_info
[ INFO] [1534836435.549527445]: Advertized on topic right/image_rect_color
[ INFO] [1534836435.563973083]: Advertized on topic right/image_raw_color
[ INFO] [1534836435.564320104]: Advertized on topic right/camera_info
[ INFO] [1534836435.579492551]: Advertized on topic depth/depth_registered
[ INFO] [1534836435.579904817]: Advertized on topic imu
Compute Mode: MODE_AUTO_DETECT
Detect CUDA not support
Load libmynteye_features.so
Load libmynteye_features.so done
Warning: Failed to get the camera's infomation.
Warning: sn is empty.
Warning: Failed to get the camera's infomation.
Error: The version is empty. (camera infomation is unexpected)
Gain: min=0, max=48, default=24
Brightness: min=0, max=240, default=120
Contrast: min=0, max=255, default=127
AutoExposure: min=0, max=1, default=0
Maxgain: min=0, max=48, default=48
MaxExposureTime: min=0, max=240, default=240
Desired brightness: min=0, max=255, default=192
ImuFrequency: min=100, max=500, default=200
IrControl: min=0, max=160, default=0
Hdr: min=0, max=1, default=0
FrameRate: min=10, max=60, default=25

LSXiang, 2018-08-21:
@WwYyFan I am not sure. You can consult AE or create a new issue.

ziyanghong, 2018-08-21:
Hi @LSXiang I am also facing the same issue, have you fixed it?

LSXiang, 2018-08-22:
@ziyanghong Not yet. U can see my answer on this issue and it might help.

indigomega, 2018-08-28:
Just received my mynt eye and came across the same problem in ROS. However, the topics are published properly. Not entirely sure why. For the future, the warning messages should be suppressed.

closed 2018-12-17 by KalmanXiong