slightech/MYNT-EYE-S-SDK #10: Unusable DEPTH from ROS wrapper

fiorano10, 2018-07-25:
I'm using the latest SDK, firmware but I'm not getting usable depth map from the camera before or after calibrating it. 
[Depth map](https://drive.google.com/open?id=1ZydsNqQlxSkRTl23yQgZHD8-A-OL9wKc)
[Image params](https://drive.google.com/open?id=1-4IK6yx-B5J2VuhfPXHalL0e09lFo-X3)

yangruixuan, 2018-07-28:
The SDK2 use the CPU depth map for default. If you have CUDA support, you can use the plug-in to enable the CUDA depth map.

fiorano10, 2018-07-28:
Even if I’m using the CPU, it should be able to provide reasonable depth map. Even the default disparity stream operates at a max of ~8 Hz. 

If my calibration is off, the disparity map should be off but it’s just the depth map.

tgreier, 2018-08-14:
Could you please explain what you mean by 'use the plug-in to enable the CUDA depth map'?

Osenberg-Q, 2018-08-17:
@tgreier Plugin have been updated on Jetson TX2 with CUDA 9 and OpenCV3.3.1

closed 2018-11-22 by yangruixuan

Sciroccogti, 2020-02-08:
Even now I'm using suitable plugin, the depth is still unusable.

The environment is Jetson TX2 with CUDA 10.0 and OpenCV 3.3.1.