slightech/MYNT-EYE-S-SDK #12: Disparity frequency lower than camera FPS

fiorano10, 2018-07-25:
When I run the image stream @10Hz, the disparity runs barely @5Hz. I'm running a 6th Gen i7 NUC, Ubuntu 16.04, and the new firmware and SDK.

tgreier, 2018-08-14:
I experienced the same issue on Jetson TX2. Video stream at 20FPS, disparity runs at 4FPS.
I see in the SDK source that cv::StereoSGBM is being used to compute the disparity map. I was led to believe that the MYNT EYE performed its disparity analysis onboard. Why is this being done in OpenCV? And, why no option for using a GPU, if the analysis is not being performed on-board the MYNT EYE?

fiorano10, 2018-08-14:
@tgreier I think they're not downsampling the image which could also be causing this. I wrote my downsampler, and then using stereo_image_proc to get the disparity image.

closed 2018-12-17 by KalmanXiong