slightech/MYNT-EYE-S-SDK #13: Upgrading firmware on Ubuntu

Schroth94, 2018-08-08:
Hello,
today I installed the MYNT-EYE-SDK-2 following the [instruction manual](https://slightech.github.io/MYNT-EYE-SDK-2-Guide/src/sdk/install_ubuntu.html), beginning with `make init` then `make install` and concluding with `make samples`. However, the device can not be read. When using `./samples/_output/bin/api/camera_a` or `./samples/_output/bin/device/camera_d`, it gives the following output:
```
I0808 15:08:42.201573  9577 utils.cc:26] Detecting MYNT EYE devices
W0808 15:08:42.545740  9577 uvc-v4l2.cc:236] xu_control_query failed error 2, No such file or directory
W0808 15:08:42.545825  9577 channels.cc:555] GetFiles failed
F0808 15:08:42.545861  9577 device.cc:546] Read device infos failed. Please upgrade your firmware to the latest version.
*** Check failure stack trace: ***
    @     0x7f2b27ab711d  google::LogMessage::Fail()
    @     0x7f2b27ab95da  google::LogMessage::SendToLog()
    @     0x7f2b27ab6b2d  google::LogMessage::Flush()
    @     0x7f2b27aba579  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f2b27a8e24c  mynteye::Device::ReadAllInfos()
    @     0x7f2b27a8e865  mynteye::Device::Device()
    @     0x7f2b27a94ccf  mynteye::StandardDevice::StandardDevice()
    @     0x7f2b27a8c452  mynteye::Device::Create()
    @     0x7f2b27a88102  mynteye::Context::Context()
    @     0x7f2b27a87316  mynteye::device::select()
    @     0x7f2b27a96145  mynteye::API::Create()
    @     0x5613f5ebe8b5  main
    @     0x7f2b266dfb97  __libc_start_main
    @     0x5613f5ec0dfa  _start
    @              (nil)  (unknown)
Aborted (core dumped)
```
It tells me to upgrade the firmware to the latest version. However, the instruction manual only provides firmware for Windows. Is there any workaround? Maybe the error comes from somewhere else. Any suggestions? I set my `$MYNTEYE_SDK_ROOT` manually to the right direction. Did I miss some important step? The device is plugged into a USB3.0 port and I have waited several time in case there are internal calibrations happening. When executing `v4l2-ctl`, it gives the following output:
```
v4l2-ctl --list-devices
MYNTEYE (usb-0000:00:14.0-1):
	/dev/video0
```
Thank you for your help.

Schroth94, 2018-08-09:
I tried to install the firmware from the `MYNTEYE_S_2.0.0_rc.img` located in MYNTEYE-BOX. However, if I mount the *.img file, it shows no files or anything. I tried using the commandline, and with a tool Furius Mount Tool.
Is it even the correct approach to get the firmware from this *.img file?

yangruixuan, 2018-08-10:
Hi please follow https://slightech.github.io/MYNT-EYE-SDK-2-Guide/src/firmware/contents.html to upgrade firmware. You need install MYNT EYE Tool in Windows and upgrade img with it

yangruixuan, 2018-08-16:
You have to upgrade in Windows

closed 2018-08-16 by yangruixuan