slightech/MYNT-EYE-S-SDK #15: Too much CPU costs.

jingedawang, 2018-08-17:
Running `mynt_eye_ros_wrapper` on my machine costs too much CPU resources.

Using default settings, CPU costs is nearly 220% (4 core):
```
top - 22:14:19 up  2:00,  4 users,  load average: 2.36, 2.45, 3.09
Tasks: 216 total,   2 running, 214 sleeping,   0 stopped,   0 zombie
%Cpu(s): 54.4 us,  0.8 sy,  0.0 ni, 44.7 id,  0.0 wa,  0.0 hi,  0.1 si,  0.0 st
KiB Mem:   3966772 total,  1043036 used,  2923736 free,    70116 buffers
KiB Swap:  4113404 total,        0 used,  4113404 free.   407804 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                 
27930 ares      20   0 1722596  83536  30568 R 221.5  2.1   3:00.40 mynteye_wrapper                                         
    1 root      20   0   33908   4500   2760 S   0.0  0.1   0:01.16 init                                                    
    2 root      20   0       0      0      0 S   0.0  0.0   0:00.00 kthreadd                                                
    3 root      20   0       0      0      0 S   0.0  0.0   0:01.53 ksoftirqd/0                                             
    5 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 kworker/0:0H                                            
    7 root      20   0       0      0      0 S   0.0  0.0   0:05.91 rcu_sched                                               
    8 root      20   0       0      0      0 S   0.0  0.0   0:00.00 rcu_bh                                                  
    9 root      rt   0       0      0      0 S   0.0  0.0   0:00.02 migration/0                                             
```

Hardware configuration:
**Intel(R) Core(TM) i5-5200U CPU @ 2.20GHz
4GB Memory**

Does it really need so much CPU resources?

joinAero, 2018-09-11:
Could disable unused synthetic streams in mynteye.launch.

```
  <!-- stream toggles -->

  <arg name="enable_left_rect" default="false" />
  <arg name="enable_right_rect" default="false" />
  <arg name="enable_disparity" default="false" />
  <arg name="enable_disparity_norm" default="false" />
  <arg name="enable_points" default="false" />
  <arg name="enable_depth" default="false" />
```

left_rect, ..., depth streams are generated by the sdk, it will cost CPU resources.

closed 2018-10-03 by yangruixuan

BernardinD, 2019-05-17:
Adding these lines to mynteye.launch does nothing for me.

jackmead515, 2020-09-22:
Don't think this is a closed issue. No working resolution. Editing the get_depth sample like so:
```
api->DisableStreamData(Stream::RIGHT);
api->DisableStreamData(Stream::LEFT_RECTIFIED);
api->DisableStreamData(Stream::RIGHT_RECTIFIED);
api->DisableStreamData(Stream::DISPARITY);
api->DisableStreamData(Stream::DISPARITY_NORMALIZED);
api->DisableStreamData(Stream::POINTS);

api->EnableStreamData(Stream::LEFT);
api->EnableStreamData(Stream::DEPTH);
```

Gives me:
```
E/processor.cc:337 Failed to get stream data of Stream::DEPTH, unsupported or disabled
```

I can't disable any stream data and get depth.

ramilmsh, 2021-03-05:
@jackmead515 If this is still relevant to you, I have looked at the source code and they simply use opencv SGBM to calculate disparity map. Which means, there is quite a bit of flexibility to be had here, as in you can change the settings (specifically the mode setting, which changes the algorithm and, possibly, down-sample the inputs). This will likely impact quality of the depth map, but it might save you some CPU time.