slightech/MYNT-EYE-S-SDK #21: Device is not supported by MYNT EYE

BrianTan34, 2018-11-17:
Hello,

I am testing `S1030-IR`, connecting to laptop 

1. w/ stock usb cable, shipped with package
2. w/o hub
3. tried on all usb ports
4. w/ laptop power cord attached/not attached

Still see below message

```
I/utils.cc:28 Detecting MYNT EYE devices
E/context.cc:38 Device is not supported by MYNT EYE.
camera_a(13156,0x7fff95b36380) malloc: *** error for object 0x7fa74340d518: incorrect checksum for freed object - object was probably modified after being freed.
*** set a breakpoint in malloc_error_break to debug
[1]    13156 abort      ./camera_a
```

Thank you

BrianTan34, 2018-11-19:
Hello, is it mandatory to upgrade to 2.x firmware? I don't have access to window machines.

KalmanXiong, 2018-11-20:
Need to updrade to 2.x firmware

BrianTan34, 2018-11-20:
Okay, managed to upgrade firmware to 2.2.1. Referred to [this page](http://doc.myntai.com/resource/sdk/mynt-eye-sdk-guide-2.0.1-rc0-html-en/mynt-eye-sdk-guide-2.0.1-rc0-html-en/src/firmware/upgrade.html) for updating firmware. When I use myntai tool, it connects to the camera and can stream videos. 

But back to previous machine, it still fails with same message.

KalmanXiong, 2018-11-22:
1,Confirm is the latest SDK
2, Run ./tools/_output/bin/writer/save_all_infos and send me the screenshots.
3,I'd like to see a picture of mynteye to confirm the hardware version
4,This page may help you. https://slightech.github.io/MYNT-EYE-S-SDK-Guide/src/firmware/from_v1.html

BrianTan34, 2018-11-22:
Not sure if it is latest SDK. I basically pulled down code from [this repo](https://github.com/slightech/MYNT-EYE-S-SDK).

`./tools/_output/bin/writer/save_all_infos

[image]

```
Product model: S1030-IR
Manufacturer: MYNT AI
Product ID: xxxx
```

yangruixuan, 2018-11-22:
HI @BrianTan34 What is your OS and hardware?

BrianTan34, 2018-11-22:
Hi @yangruixuan thanks for your reply. OS X. 10.13. In terms of hardware, I guess you mean camera hardware?

yangruixuan, 2018-11-23:
OK, Mac OS is not support by SDK 2.0 yet. We are working on that...

BrianTan34, 2018-11-23:
Is SDK 1.0 still available? or it is discontinued, we just can't use it with OSX by any means?

yangruixuan, 2018-11-24:
SDK 1.x is not recommend and not support any more(you can find it here: https://github.com/slightech/MYNT-EYE-SDK ). We are developing SDK 2.0 OSX version. Maybe can be finished in Dec.

closed 2018-12-27 by KalmanXiong