slightech/MYNT-EYE-S-SDK #24: provide api function for retrieving current hardware clock

v01d, 2018-11-22:
It would be really useful to obtain current camera clock since the current approach is to translate the image/imu-segments time to ROS time using the time of reception. Both messages are delayed w.r.t. to the actual sensing time (image needs to be exposed and transferred and IMU segment accumulates several readings), so estimating the ROS time corresponding to current hardware time in this way is not correct. In fact, Imu messages are currently published with future times since the mapping is initialized from the first reading of the packet (assuming the Imu message comes first) and then the remaining ones already received have future timestamps.

If this is not feasible, please at least inform of imu/image transmision delays so that we can compensate for that when making the ROS-to-hardware time mapping.

KalmanXiong, 2018-12-27:
Our latest release has modified this issue

closed 2018-12-27 by KalmanXiong

bxc237, 2019-03-22:
@KalmanXiong Hi, I have compiled the ROS wrapper in the master branch, but still found that the imu/image timestamp was from the future with respect to the ros::time::now(). What is the time you guys stamped to the message? Does the time have relation with ros time or it is a pure hardware time from the sensor? Thanks.