slightech/MYNT-EYE-S-SDK #25: MYNTEYE S - IMU message drops

mfehr, 2018-11-23:
We just received a new mynteye s sensor after initially buying an earlier version. With the previous sensor we observed and reported IMU message drops (See https://github.com/slightech/MYNT-EYE-SDK/issues/22).

The new sensor has these issues as well, after restamping the bag (writing the messages into the bag based on the timestamp of the ros headers):

Original bag:
[image]
Restamped bag:
[image]

Here is a plot showing the drops:

[image]

There are IMU messages missing, 1 to 5 at a time, this can also be seen in the ROS sequence number of the messages.

After reporting this before in the previous sensor the issue was closed and the recommended firmware update failed. Now with a brand-new sensor and the new SDK we experience the same issue. Could you help us resolve this issue?

mfehr, 2018-11-24:
I had another look at this issue and also the ros publisher, because we suspected it might just be ROS dropping messages due to overloading the queue. Especially the fact that the sequence numbers actually show a drop as well makes it more likely to be an issue in the ROS node/queues than the sensor itself. This time I used just the mynteye.launch file (not display.launch which has rviz subscribe to some heavier topics). In the launch file all the additional streams are disabled.
```xml
  <arg name="enable_left_rect" default="false" />
  <arg name="enable_right_rect" default="false" />
  <arg name="enable_disparity" default="false" />
  <arg name="enable_disparity_norm" default="false" />
  <arg name="enable_points" default="false" />
  <arg name="enable_depth" default="false" />
```
 Looks like this helps and the drops happen very rarely now, but they still do (note that this rosbag is a lot longer):

[image]

Any ideas of how to get rid of the remaining drops? I will at least need to be able to record images, imu and ideally depth at the same time.

yangruixuan, 2018-11-24:
Hi @mfehr Can you give some info like your MYNT EYE Model, Firmware version, SDK version, Hardware info etc. Thanks

mfehr, 2018-11-25:
Hi @yangruixuan 
Sensor: MYNT-EYE-S1030 serial: 05071F0700090721 firmware version: 2.2
SDK: this repo, master branch
PC: Xeon CPU E3-1505M v5 @ 2.80GHz, with SSD
OS: Ubuntu 18.04, ros melodic

yangruixuan, 2018-11-25:
I think you are right. The issue may caused by queue problem. You can try this change
    pub_imu_ = nh_.advertise<sensor_msgs::Imu>(imu_topic, 1);
to
    pub_imu_ = nh_.advertise<sensor_msgs::Imu>(imu_topic, 100);
in MYNT-EYE-S-SDK/wrappers/ros/src/mynt_eye_ros_wrapper/src/wrapper_nodelet.cc

closed 2018-12-27 by KalmanXiong