slightech/MYNT-EYE-S-SDK #28: Cant run sample code as instructed

TrungKhoaLe, 2018-12-14:
I try to run the sample code as instructed but nothing happens as described. I sent email to technical support and have got no comment. Now my business fucked up.

KalmanXiong, 2018-12-17:
I want to know more details for your problem

TrungKhoaLe, 2018-12-17:
I myself solved the problem. My problem is that the MYNT EYE camera has not been recognized automatically by my laptop as mentioned in guideline. 
> Unpack your MYNT EYE and plug the camera in a USB 3.0 port. The MYNT EYE is a UVC compliant camera so it should be automatically recognized by your computer.

Then I followed this [https://slightech.github.io/MYNT-EYE-SDK/upgrade_firmware.html](url) in "Other Instructions" part in order to install the driver. Problem solved.

closed 2018-12-17 by TrungKhoaLe