slightech/MYNT-EYE-S-SDK #31: Segmentation Fault whenn running mynteye_wrapper_node

nicotho, 2018-12-26:
Hi,

When starting the mynteye ros node under Ros Melodic the node stops with a segmentation fault after advertizing service get_info. I have compiled the SDK v2 and my camera config is 

`device_name: MYNT-EYE-S1030
serial_number: "0507151500090721"
firmware_version: "2.2"
hardware_version: "2.0"
spec_version: "1.0"
lens_type: "0000"
imu_type: "0000"
nominal_baseline: 120
`

After some ROSINFO debugging I found out that the node crashes when invoking 
` api_->Start(Source::ALL);`
in the function `  void publishTopics() ` of wrapper_nodelet.cc

Thank you for your support.
Nicolas

KalmanXiong, 2018-12-27:
When you run samples output bin\ api\ camera_a, do you have similar problems?

KalmanXiong, 2019-01-17:
I'm sorry I didn't notice your reply before. Do you have any problem under ROS now?

tiralonghipol, 2019-01-18:
@KalmanXiong  @nicotho 
any news on this? kinda have the same problem!!

nicotho, 2019-01-22:
@KalmanXiong 
Hi,

Unfortunately my problem persists. How do we proceed? Is there anything I prepare in order for you to Debug the situation?
Thanks,
Nicolas

huanglilong, 2021-10-16:
i find coredump here: src/mynteye/api/processor/disparity_processor.cc: sgbm_matcher->setPreFilterCap(63);