slightech/MYNT-EYE-S-SDK #35: Python 接口不对应

duchengyao, 2019-01-18:
3.2.0 版本更新了[这行](https://github.com/slightech/MYNT-EYE-S-SDK/blob/master/include/mynteye/api/api.h#L95)

然而[python wrapper](https://github.com/slightech/MYNT-EYE-S-SDK/blob/master/wrappers/python/src/mynteye_py.cc#L113)没有更新

导致错误
```

/Users/duchengyao/project/SLAM/MYNT-EYE-S-SDK-2.3.0/wrappers/python/src/mynteye_py.cc:113:54: error: no matching constructor for initialization of
      'mynteye::API'
  explicit APIWrap(std::shared_ptr<Device> device) : API(device) {}
                                                     ^   ~~~~~~
/usr/local/include/mynteye/api/api.h:88:19: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from
      'std::shared_ptr<Device>' to 'const mynteye::API' for 1st argument
class MYNTEYE_API API {
                  ^
/usr/local/include/mynteye/api/api.h:95:12: note: candidate constructor not viable: requires 2 arguments, but 1 was provided
  explicit API(std::shared_ptr<Device> device, CalibrationModel calib_model);
  
```

harjeb, 2019-02-11:
感谢指出错误，python wrapper在2.3.0版本暂未做支持，将在之后的版本修复这个问题

closed 2019-02-11 by harjeb

IamWangYunKai, 2019-11-17:
Compiled boost with python3.7 and found missing numpy-opencv-converter, finally it comes to this problem :joy: However, it seems python wrapper cannot work in recent months. When the newest version with python wrapper will come ?