slightech/MYNT-EYE-S-SDK #37: Can't run SDK on windows 10

BrianTan34, 2019-02-02:
Hello I am following tutorial of SDK setup on windows 10.

https://slightech.github.io/MYNT-EYE-S-SDK-Guide/src/sdk/exe_install_windows.html

However when I run `get_stereo` and any other executable, the console shows 

`Detecting MYNT EYE devices` 

then quit.

However windows system notifies that it detected myntai device after SDK installation and system reboot.

Please advise, thank you!

BrianTan34, 2019-02-03:
Thank you for your reply. Under VM windows 10. Port was selected as USB 3.0. I am using same machine (native OSX) running `uvc` sample, getting output.

I am also trying to understand the frame output. Is it stereo images overlaid together? Let me know if wants to open another issue.

[image]

BrianTan34, 2019-02-03:
Tried to do below request on OSX, however,  
```
 int main(int argc, char *argv[]) {
     glog_init _(argc, argv);

     auto &&device = device::select();
     if (!device) return 1;

     bool ok;
     auto &&request = device::select_request(device, &ok);
     if (!ok) return 1;
```
Then I encountered this error:

[image]

So I decide to update firmware under windows VM with latest 2.2.2 ver firmware

But encountered this error

[image]

harjeb, 2019-02-11:
hi @BrianTan34 ，SDK in mac OS just support `uvc` now, please use Windows 10 or Ubuntu to get all the features(not support VM also)

BrianTan34, 2019-02-17:
Thank you. Will close then.

closed 2019-02-17 by BrianTan34