slightech/MYNT-EYE-S-SDK #40: Jetson TK1 crash when running camera_a

g-spacewhale, 2019-02-21:
I use a  MYNT-EYE-S1030 with the newest firmware. 
[image]

I installed this SDK on the Nvidia Jetson TK1; when running the camera_a  program, the TK1 freezes and then restarts after ~ 30 sec. Debugging the code, shows, that the program dies when calling ioctl() following the execution of this part of the _start_capture()_ function in uvc-v4l2.cc:

[image]

ioctl() returns -1 with an errno of 16 (device or resource busy). The previous value of errno after calling ioctl is 25 (Inappropriate ioctl for device).

The TK1 runs  Linux version 3.10.40-ga7da876 (buildbrain@mobile-u64-234) (gcc version 4.5.3 (crosstool-NG hg_unknown@20110628.165246) )

Connecting the MYNT EYE to a PC running ubuntu 14.04 and using this SDK to run camera_a worked with no problem. 

Can anyone explain, why this happens or/and assist in resolving this issue?
I will gladly supply any further necessary information about the setup.

best wishes

g-spacewhale, 2019-02-22:
Solution:
The respective ioctl() call originating from the `_startup_capture()_` with the `request = 0x40045612` (in hex) is called twice by design, even if the first ioctl call returns no error:

the original code (lines 334 - 343 in vim uvc-v4l2.cc )

```

335     for (int i = 0; i < 10; ++i) {
336       if (xioctl(fd, VIDIOC_STREAMON, &type) < 0) {
337         std::this_thread::sleep_for(std::chrono::milliseconds(100));
338       }
339     }
340     if (xioctl(fd, VIDIOC_STREAMON, &type) < 0)
341       LOG_ERROR(FATAL, "VIDIOC_STREAMON");
342 
343     is_capturing = true;
```
This does not represent a problem on the i86 64 bit, the call returns 0 error on both occasions (call 336 and 340. 

The change I made in the code to allow for repeatable execution, calling the ioctl only once:

```
335     for (int i = 0; i < 10; ++i) {
336       if (xioctl(fd, VIDIOC_STREAMON, &type) < 0) {
337         std::this_thread::sleep_for(std::chrono::milliseconds(100));
338       }
339       else
340       {
341         is_capturing = true;
342         return;
343       }
344     }
```

Why is this not the default way of handling the call?

I don't know yet, why this repeated call not only breaks the program but breaks the entire kernel completely.

joinAero, 2019-03-07:
Thanks, RD will ensure `xioctl VIDIOC_STREAMON` success only once in next version `2.3.4`.

joinAero, 2019-03-07:
Pushed this fix to `develop` branch

closed 2019-03-23 by harjeb