slightech/MYNT-EYE-S-SDK #41: make ros error:  file failed to open for writing

letsrocking, 2019-02-22:
sudo make install is OK，but

robot@robot-desktop:~/soft/MYNT-EYE-S-SDK$ make ros
Make build
CD: ./_build
CMake Error at cmake/TargetArch.cmake:131 (file):
  file failed to open for writing (Permission denied):

    /home/robot/soft/MYNT-EYE-S-SDK/_build/arch.c
Call Stack (most recent call first):
  cmake/Common.cmake:50 (target_architecture)
  CMakeLists.txt:19 (include)

CMake Error at cmake/TargetArch.cmake:143 (try_run):
  Failed to open

    /home/robot/soft/MYNT-EYE-S-SDK/_build/CMakeFiles/CMakeTmp/CMakeLists.txt

  Permission denied
Call Stack (most recent call first):
  cmake/Common.cmake:50 (target_architecture)
  CMakeLists.txt:19 (include)

-- HOST_ARCH: unknown
-- 
-- Platform:
--   HOST_OS: Linux
--   HOST_NAME: Ubuntu
--   HOST_ARCH: unknown
--   HOST_COMPILER: GNU
--     COMPILER_VERSION: 5.4.0
--     COMPILER_VERSION_MAJOR: 5
--     COMPILER_VERSION_MINOR: 4
--     COMPILER_VERSION_PATCH: 0
--     COMPILER_VERSION_TWEAK: 
--   OpenCV_VERSION: 3.4.5
--     OpenCV_VERSION_MAJOR: 3
--     OpenCV_VERSION_MINOR: 4
--     OpenCV_VERSION_PATCH: 5
--     OpenCV_VERSION_TWEAK: 0
--     OpenCV_VERSION_STATUS: 
--     OpenCV_WITH_WORLD: 
--   MYNTEYE_VERSION: 2.3.0
--     MYNTEYE_VERSION_MAJOR: 2
--     MYNTEYE_VERSION_MINOR: 3
--     MYNTEYE_VERSION_PATCH: 0
--     MYNTEYE_VERSION_TWEAK: 
-- 
-- Options:
--   WITH_API: ON
--     OpenCV: YES
--     OpenCV_VERSION: 3.4.5
--     OpenCV_WORLD: NO
--   WITH_DEVICE_INFO_REQUIRED: ON
--   WITH_CAM_MODELS: OFF
--   WITH_BOOST: ON
--     Boost: YES
--     Boost_VERSION: 1.58.0
--   WITH_GLOG: OFF
-- 
-- Features:
--   Filesystem: boost
-- 
-- Using flag -std=c++11
-- C_FLAGS: -Wall -Wextra -march=native
-- CXX_FLAGS: -Wall -Wextra -std=c++11 -march=native
-- Namespace: mynteye
CMake Error: Could not open file for write in copy operation /home/robot/soft/MYNT-EYE-S-SDK/_build/include/mynteye/mynteye.h.tmp
CMake Error: : System Error: Permission denied
CMake Error at CMakeLists.txt:81 (configure_file):
  configure_file Problem configuring file

-- RPATH: /usr/local/lib
CMake Error: Could not open file for write in copy operation /home/robot/soft/MYNT-EYE-S-SDK/_build/mynteye-config.cmake.tmp
CMake Error: : System Error: Permission denied
CMake Error at /usr/share/cmake-3.5/Modules/CMakePackageConfigHelpers.cmake:323 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:370 (configure_package_config_file)

CMake Error: Could not open file for write in copy operation /home/robot/soft/MYNT-EYE-S-SDK/_build/mynteye-config-version.cmake.tmp
CMake Error: : System Error: Permission denied
CMake Error at /usr/share/cmake-3.5/Modules/WriteBasicConfigVersionFile.cmake:59 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CMakePackageConfigHelpers.cmake:220 (write_basic_config_version_file)
  CMakeLists.txt:376 (write_basic_package_version_file)

CMake Error: Could not open file for write in copy operation /home/robot/soft/MYNT-EYE-S-SDK/_build/build.info.tmp
CMake Error: : System Error: Permission denied
CMake Error at CMakeLists.txt:408 (configure_file):
  configure_file Problem configuring file

-- Configuring incomplete, errors occurred!
See also "/home/robot/soft/MYNT-EYE-S-SDK/_build/CMakeFiles/CMakeOutput.log".
See also "/home/robot/soft/MYNT-EYE-S-SDK/_build/CMakeFiles/CMakeError.log".
CMake Error: Cannot open file for write: /home/robot/soft/MYNT-EYE-S-SDK/_build/CMakeCache.txt.tmp
CMake Error: : System Error: Permission denied
CMake Error: Unable to open cache file for save. /home/robot/soft/MYNT-EYE-S-SDK/_build/CMakeCache.txt
CMake Error: : System Error: Permission denied
Makefile:119: recipe for target 'build' failed
make: *** [build] Error 1

letsrocking, 2019-02-22:
robot@robot-desktop:~/soft/MYNT-EYE-S-SDK/_build$ ls -l
总用量 116
-rw-r--r-- 1 root root  2292 2月  22 15:54 arch.c
-rw-rw-r-- 1 root root   600 2月  22 15:54 build.info
-rw-r--r-- 1 root root 21334 2月  22 15:54 CMakeCache.txt
drwxr-xr-x 6 root root  4096 2月  22 15:55 CMakeFiles
-rw-r--r-- 1 root root 13578 2月  22 15:54 cmake_install.cmake
drwxr-xr-x 3 root root  4096 2月  22 15:54 include
-rw-r--r-- 1 root root  1384 2月  22 15:55 install_manifest.txt
-rw-r--r-- 1 root root 47715 2月  22 15:54 Makefile
-rw-rw-r-- 1 root root  1386 2月  22 15:54 mynteye-config.cmake
-rw-r--r-- 1 root root  1607 2月  22 15:54 mynteye-config-version.cmake

harjeb, 2019-03-07:
Did you use "sudo git clone" ? Try “git clone”

freeyyc, 2019-09-18:
要使用sudo make ros
不过我使用了sudo make ros 又报了另一个错误  catkin_make 未找到命令