slightech/MYNT-EYE-S-SDK #42: 2.3.0: Read device infos failed.

h4x0rz, 2019-03-04:
Upgraded firmware multiple times to 2.3.0 and updated SDK to 2.3.2 on Ubuntu 16.04 yet getting "Read device infos failed. Upgrade firmware" error messaged.
So I switched to windows and got the new Windows SDK and tried the samples - works perfectly!

Any errors with the 2.3.2 on Linux?

harjeb, 2019-03-07:
hi,  pu(Processing Unit) and xu(Extension Unit）are used by UVC, These errors means usb/uvc driver not work well or usb poor contact.

jameswalmsley, 2019-04-25:
Hi @h4x0rz I believe this is due to an incorrect kernel version in the newer versions of Ubuntu.
Essentially V4L2 has changed and now instead of a single /dev/video device, V4L creates 2 for each camera.

See if you can install kernel version 4.14 (4.15 is when V4L changed).

I'm trying to get this to work myself, I'll let you know if I succeed.

jameswalmsley, 2019-04-25:
@harjeb See:: https://github.com/IntelRealSense/librealsense/issues/1514

Realsense had the same issue.

jameswalmsley, 2019-04-25:
Works great:
See: https://wiki.ubuntu.com/Kernel/MainlineBuilds

Download:
https://kernel.ubuntu.com/~kernel-ppa/mainline/v4.14.113/

* linux-image-unsigned-4.14.113-0414113-generic_4.14.113-0414113.201904200930_amd64.deb
* linux-modules-4.14.113-0414113-generic_4.14.113-0414113.201904200930_amd64.deb

    sudo dpkg -i *.deb

Then reboot into the correct kernel using the grub menu on reboot.
(Comment out GRUB_TIMEOUT=0 in /etc/default/grub and the run sudo update-grub before rebooting).

Hope this helps people get up and running for now.
@harjeb  @mynteye hope you are aware of this and how to solve it.

harjeb, 2019-04-29:
@jameswalmsley Yes, looks like you are correct, thanks for reporting this,  @h4x0rz please use kernel version <=4.15, we'll solve this issue sooner.

closed 2019-04-29 by harjeb

bergercookie, 2019-05-22:
Hey @harjeb,

Is this issue solved? If so could you point me to the branch/commit that I should be using to get this fix?

harjeb, 2019-05-23:
@bergercookie hi, make sure firmware version is 2.4.0 ，and kernel version <= 4.15.  xu_control_query and pu_control_range errors related to USB or UVC driver, not SDK.  Please have a try on other USB ports or computers

bergercookie, 2019-05-24:
I understand. It's just a bit confusing though closing the issue since there is no actual fix for kernel >= 4.15.

agrathwohl, 2019-06-06:
Hey @bergercookie,

Do I understand correctly, that this issue makes the MYNT-EYE-S SDK unusable on any Linux whose kernel is >=4.15? Or is this just an issue with Ubuntu?

bergercookie, 2019-06-06:
Hey @agrathwohl,

Correct.