slightech/MYNT-EYE-S-SDK #55: POINT CLOUD, MYNT EYE S

logindiamant, 2019-07-16:
Dear collegues!

We want to get high quality point cloud from MYNT EYE S1030 cam, without empty undefined areas as shown below
Please, help us to solve problem

[image]
[image]
[image]
[image]
[image]

harjeb, 2019-07-17:
Hi, there are two disparity computing method in SDK，default is bm and the market picture use sgbm. 
you can follow https://mynt-eye-s-sdk.readthedocs.io/en/latest/src/sdk/data/get_disparity.html to change. Or you can modify the config [file](https://github.com/slightech/MYNT-EYE-S-SDK/blob/master/wrappers/ros/src/mynt_eye_ros_wrapper/config/process/process_config.yaml) to use it in ROS

logindiamant, 2019-07-17:
Thanks for your answer, harjeb!

I change algorithm to sgbm. Now we have a good point clouds!
[image]
[image]
[image]

First question is, if I change
api->SetOptionValue(Option::IR_CONTROL, 160);
from infrared.cc, does this change saved in cam, and ready to apply in ros wraper?

Second question is, does MYNT EYE have methods to thin out a fake points?
Same as in RealSense
https://github.com/IntelRealSense/librealsense/blob/master/src/proc/decimation-filter.cpp

[image]
[image]

harjeb, 2019-07-18:
Hi， For S1030 device you need modify this  [file](https://github.com/slightech/MYNT-EYE-S-SDK/blob/cdae311e99574516f151a5e96bd70faf26696326/wrappers/ros/src/mynt_eye_ros_wrapper/config/device/standard.yaml#L38) to change the IR strength. Here is document about use ros https://mynt-eye-s-sdk.readthedocs.io/en/latest/src/wrapper/ros.html#wrapper-ros.
And for now, we don't have methods to thin out a fake points

logindiamant, 2019-07-18:
Thanks, harjeb
Please tel us, how to change/reduce resolution with sgbm?
Or other methods to speed up calculations. In Jetson TX2 we have only 3-4Hz point cloud
I try to change standard/request_index: 0 to 1 in standard.yaml, but device is crashed in ros

harjeb, 2019-07-19:
Hi，this crash issue has fixed in develop branch, you can switch to this branch. And sgbm do spend much more resources than bm algorithm, and this is the config [file ](https://github.com/slightech/MYNT-EYE-S-SDK/blob/develop/config/disparity_config.yaml)

logindiamant, 2019-07-19:
We have already speed up calculations to 30Hz in NUC BLKNUC7I7DNK2E
# standard/frame_rate range: {10,15,20,25,30,35,40,45,50,55,60}
standard/frame_rate: 30
# MYNTEYE-S1030, Reslution: 376x240, Format: YUYV
standard/request_index: 1
develop branch
Thanks, harjeb!

logindiamant, 2019-07-19:
We plan to change our multi-cam RealSense platform to multi-cam MYNT EYE
Because we have a big problem, usb hub Orico destroy 4 RealSenses...

Please, tel us what usb hubs you recomended for plug in 2 and 4 MYNT EYE S1030

closed 2019-07-23 by logindiamant