slightech/MYNT-EYE-S-SDK #57: cant set frame rate other than 25

phil333, 2019-07-23:
I ve been trying to set the frame rate to 5fps, but if i change the config file, it does not affect the frame rate of the camera, even if i use other values like 10 or 15. I can change other values in the config file, like the IR projection strength, but the fps is broken.

harjeb, 2019-07-24:
If you use S1030, just modify "-1" to “10” in this [line](https://github.com/slightech/MYNT-EYE-S-SDK/blob/cdae311e99574516f151a5e96bd70faf26696326/wrappers/ros/src/mynt_eye_ros_wrapper/config/device/standard.yaml#L14)

phil333, 2019-07-25:
Ok, somehow works now. Is there any way to go lower than 10? 5fps is what we are looking for

harjeb, 2019-08-05:
Sorry, 10 fps is lowest frame rate now