slightech/MYNT-EYE-S-SDK #61: Python wrapper not compiling (incorrect function signature)

rosshemsley, 2019-08-13:
The Python wrappers fail to compile for me.
It looks like the api was changed but the Python wrapper has not been updated to take account of it:

* The API class is instantiated with a single "device" ptr, but the API as declared actually requires the calibration model parameter:

https://github.com/slightech/MYNT-EYE-S-SDK/blob/master/include/mynteye/api/api.h#L98
vs.
https://github.com/slightech/MYNT-EYE-S-SDK/blob/master/wrappers/python/src/mynteye_py.cc#L113

* I also get some boost errors but I didn't dive into that.

harjeb, 2019-08-13:
Hi, I think you're right and python wrapper doesn't update after v2.0, we'll support python api later.

pirazor, 2019-08-13:
Desperately in need of a fix for the python wrapper compilation; spent hours on it but could not compile, had the same problem.

alter-sachin, 2019-09-17:
Nudge.

alter-sachin, 2019-10-23:
https://github.com/alter-sachin/MYNT

I have been able to make the python wrapper working here.
Requires boost 1.71 built wiith python3.6 
This works for me. Use at your discretion

alter-sachin, 2019-10-25:
https://github.com/alter-sachin/MYNT-EYE-S-SDK

I have been able to make the wrapper working for python3 . please use if helpful.