slightech/MYNT-EYE-S-SDK #63: Sample camera_a segmentation fault with "no suited processor"

clement-joly, 2019-08-29:
I've just installed the SDK from source without problems but when I run the sample program ./samples/_output/bin/api/camera_a I get the following error message:

`~/Softwares/MYNT-EYE-S-SDK$ ./samples/_output/bin/api/camera_a
I/utils.cc:35 Detecting MYNT EYE devices
W/uvc-v4l2.cc:218 pu_control_range failed error 22, Invalid argument
W/channels.cc:806 Get PuControlInfo of Option::GAIN failed
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 00D303260009072C, firmware: 2.4
I/utils.cc:60 Only one MYNT EYE device, select index: 0
E/synthetic.cc:370 Unknow calib model type in device: CalibrationModel::KANNALA_BRANDT
I/utils.cc:93 MYNT EYE requests:
I/utils.cc:96   index: 0, request: width: 752, height: 480, format: Format::YUYV, fps: 60
I/utils.cc:96   index: 1, request: width: 376, height: 240, format: Format::YUYV, fps: 60
I/utils.cc:107 There are 2 stream requests, select index: 
0
E/synthetic.cc:139 ERROR: no suited processor for stream Stream::LEFT_RECTIFIED
E/synthetic.cc:107 Unknow calib model type in device
W/uvc-v4l2.cc:234 pu_control_query failed error 22, Invalid argument
W/channels.cc:190 Option::GAIN get value failed
I/channels.cc:136 Option::GAIN: min=0, max=0, def=0, cur=-1
I/channels.cc:136 Option::BRIGHTNESS: min=0, max=240, def=120, cur=120
I/channels.cc:136 Option::CONTRAST: min=0, max=255, def=127, cur=127
I/channels.cc:136 Option::FRAME_RATE: min=10, max=60, def=25, cur=25
I/channels.cc:136 Option::IMU_FREQUENCY: min=100, max=500, def=200, cur=500
I/channels.cc:136 Option::EXPOSURE_MODE: min=0, max=1, def=0, cur=0
I/channels.cc:136 Option::MAX_GAIN: min=0, max=48, def=48, cur=48
I/channels.cc:136 Option::MAX_EXPOSURE_TIME: min=0, max=240, def=240, cur=240
I/channels.cc:136 Option::DESIRED_BRIGHTNESS: min=0, max=255, def=192, cur=192
I/channels.cc:136 Option::IR_CONTROL: min=0, max=160, def=0, cur=0
I/channels.cc:136 Option::HDR_MODE: min=0, max=1, def=0, cur=0
I/channels.cc:136 Option::ACCELEROMETER_RANGE: min=4, max=32, def=8, cur=8
I/channels.cc:136 Option::GYROSCOPE_RANGE: min=500, max=4000, def=1000, cur=1000
E/synthetic.cc:154 ERROR: no suited processor for stream Stream::LEFT
E/synthetic.cc:154 ERROR: no suited processor for stream Stream::RIGHT
E/synthetic.cc:139 ERROR: no suited processor for stream Stream::DISPARITY_NORMALIZED
Segmentation fault (core dumped)`

The same issue happens when using the SDK installed from PPA.

My machine is a Dell G3 15-3579, Intel processor.

Any idea what's happening or how I can troubleshoot this?

harjeb, 2019-08-29:
need information about system and SDK version

clement-joly, 2019-08-30:
Here it is:
MYNT-EYE-S1030, sn: 00D303260009072C, firmware: 2.4 
The SDK version is the latest, I pulled it from Github on Wednesday.
Let me know if you need anything else.

harjeb, 2019-09-02:
This error seems like something wrong with uvc driver.  Which system you use?  Ubuntu 16.04 LTS or some else?

clement-joly, 2019-09-02:
I'm using Ubuntu 18.04

clement-joly, 2019-09-03:
If you need me to run some scripts / other code to help debugging, let me know harjeb, I'd be happy to help!

harjeb, 2019-09-05:
@clement-joly please check the kernel version， if you use kernel >4.15 ，need update to latest SDK.

YagnikThanki, 2019-09-10:
I am using 
MYNT-EYE-S2100, sn: 07C619160009071F, firmware: 1.3 with ubuntu 18.04 and kernal version 5.0.
Do i need to update firmware and latest SDK?
How can i update latest SDK?

alter-sachin, 2019-09-15:
Were you able to resolve this ?

harjeb, 2019-09-16:
@YagnikThanki you can update SDK like this：
`cd MYNT-EYE-S-SDK`
`git checkout .`
`git pull`
`make cleanall all`

harjeb, 2019-09-16:
@clement-joly Do you use PC or other demoboards? we tested on TX2 18.06 4.9.130-tegra aarch64/PC 5.0.0-27-generic X86_64 and not meet this issue

YagnikThanki, 2019-09-16:
@harjeb Thanks. How can i solve "no suited processor error" ?

alter-sachin, 2019-09-16:
I am getting a similar problem when I run 

/opt/mynt-eye-s-sdk/samples$ ./api/camera_a 

`I/utils.cc:35 Detecting MYNT EYE devices
W/uvc-v4l2.cc:218 pu_control_range failed error 22, Invalid argument
W/channels.cc:806 Get PuControlInfo of Option::GAIN failed
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 4B4E0D1100090712, firmware: 2.4
I/utils.cc:60 Only one MYNT EYE device, select index: 0
E/synthetic.cc:370 Unknow calib model type in device: CalibrationModel::KANNALA_BRANDT
2.4.2I/utils.cc:99 MYNT EYE requests:
I/utils.cc:102   index: 0, request: width: 752, height: 480, format: Format::YUYV, fps: 25
I/utils.cc:102   index: 1, request: width: 376, height: 240, format: Format::YUYV, fps: 25
I/utils.cc:113 There are 2 stream requests, select index: 
1
E/synthetic.cc:139 ERROR: no suited processor for stream Stream::LEFT_RECTIFIED
E/synthetic.cc:107 Unknow calib model type in device

W/uvc-v4l2.cc:234 pu_control_query failed error 22, Invalid argument
W/channels.cc:190 Option::GAIN get value failed
I/channels.cc:136 Option::GAIN: min=0, max=0, def=0, cur=-1
I/channels.cc:136 Option::BRIGHTNESS: min=0, max=240, def=120, cur=120
I/channels.cc:136 Option::CONTRAST: min=0, max=255, def=127, cur=127
I/channels.cc:136 Option::FRAME_RATE: min=10, max=60, def=25, cur=25
I/channels.cc:136 Option::IMU_FREQUENCY: min=100, max=500, def=200, cur=500
I/channels.cc:136 Option::EXPOSURE_MODE: min=0, max=1, def=0, cur=0
I/channels.cc:136 Option::MAX_GAIN: min=0, max=48, def=48, cur=48
I/channels.cc:136 Option::MAX_EXPOSURE_TIME: min=0, max=240, def=240, cur=240
I/channels.cc:136 Option::DESIRED_BRIGHTNESS: min=0, max=255, def=192, cur=192
I/channels.cc:136 Option::IR_CONTROL: min=0, max=160, def=0, cur=0
I/channels.cc:136 Option::HDR_MODE: min=0, max=1, def=0, cur=0
I/channels.cc:136 Option::ACCELEROMETER_RANGE: min=4, max=32, def=8, cur=8
I/channels.cc:136 Option::GYROSCOPE_RANGE: min=500, max=4000, def=1000, cur=1000
E/synthetic.cc:154 ERROR: no suited processor for stream Stream::LEFT
E/synthetic.cc:154 ERROR: no suited processor for stream Stream::RIGHT
E/synthetic.cc:139 ERROR: no suited processor for stream Stream::DISPARITY_NORMALIZED
Segmentation fault (core dumped)
`

Any help would be appreciated.

alter-sachin, 2019-09-17:
Hi, I have been able to use the older versions 2.3.9 to build and run the samples.

YagnikThanki, 2019-09-19:
@alter-sachin  Does that update include python wrapper? 
At the end I want to use python wrapper.

alter-sachin, 2019-10-10:
No, that does not include the python wrapper

alter-sachin, 2019-10-25:
https://github.com/alter-sachin/MYNT-EYE-S-SDK

I have been able to make the wrapper working for python3 . please use if helpful.

wenxini234, 2019-11-26:
I/utils.cc:35 Detecting MYNT EYE devices
W/uvc-v4l2.cc:217 pu_control_range failed error 22, Invalid argument
W/channels.cc:739 Get PuControlInfo of Option::GAIN failed
W/uvc-v4l2.cc:217 pu_control_range failed error 25, Inappropriate ioctl for device
W/channels.cc:739 Get PuControlInfo of Option::GAIN failed
W/uvc-v4l2.cc:217 pu_control_range failed error 25, Inappropriate ioctl for device
W/channels.cc:739 Get PuControlInfo of Option::BRIGHTNESS failed
W/uvc-v4l2.cc:217 pu_control_range failed error 25, Inappropriate ioctl for device
W/channels.cc:739 Get PuControlInfo of Option::CONTRAST failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:547 xu_control_range query failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:554 xu_control_range query min failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:560 xu_control_range query max failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:566 xu_control_range query def failed
W/channels.cc:750 Get XuControlInfo of Option::FRAME_RATE failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:547 xu_control_range query failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:554 xu_control_range query min failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:560 xu_control_range query max failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:566 xu_control_range query def failed
W/channels.cc:750 Get XuControlInfo of Option::IMU_FREQUENCY failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:547 xu_control_range query failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:554 xu_control_range query min failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:560 xu_control_range query max failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:566 xu_control_range query def failed
W/channels.cc:750 Get XuControlInfo of Option::EXPOSURE_MODE failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:547 xu_control_range query failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:554 xu_control_range query min failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:560 xu_control_range query max failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:566 xu_control_range query def failed
W/channels.cc:750 Get XuControlInfo of Option::MAX_GAIN failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:547 xu_control_range query failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:554 xu_control_range query min failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:560 xu_control_range query max failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:566 xu_control_range query def failed
W/channels.cc:750 Get XuControlInfo of Option::MAX_EXPOSURE_TIME failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:547 xu_control_range query failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:554 xu_control_range query min failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:560 xu_control_range query max failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:566 xu_control_range query def failed
W/channels.cc:750 Get XuControlInfo of Option::DESIRED_BRIGHTNESS failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:547 xu_control_range query failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:554 xu_control_range query min failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:560 xu_control_range query max failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
[... 10 lines cut ...]
W/channels.cc:750 Get XuControlInfo of Option::HDR_MODE failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:547 xu_control_range query failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:554 xu_control_range query min failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:560 xu_control_range query max failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:566 xu_control_range query def failed
W/channels.cc:750 Get XuControlInfo of Option::ACCELEROMETER_RANGE failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
[... cut ...]
W/uvc-v4l2.cc:560 xu_control_range query max failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/uvc-v4l2.cc:566 xu_control_range query def failed
W/channels.cc:750 Get XuControlInfo of Option::GYROSCOPE_RANGE failed
W/uvc-v4l2.cc:246 xu_control_query failed error 25, Inappropriate ioctl for device
W/channels.cc:441 GetFiles failed
F/device.cc:593 Read device infos failed. Please upgrade your firmware to the latest version.
Aborted (core dumped)

Any help would be appreciated   system: Ubuntu 18.04.1 LTS

harjeb, 2019-11-26:
@wenxini234 Hi，could attach your SDK version(use `git log`) and upgrade firmware through [doc](https://mynt-eye-s-sdk.readthedocs.io/en/latest/src/firmware/update_main_chip.html)

wenxini234, 2019-11-26:
@harjeb  name: MYNT-EYE-S1030, sn: 04081B2E00090834, firmware: 2.5
I/utils.cc:35 Detecting MYNT EYE devices
W/uvc-v4l2.cc:218 pu_control_range failed error 22, Invalid argument
W/channels.cc:1028 Get PuControlInfo of Option::GAIN failed
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 04081B2E00090834, firmware: 2.5
I/utils.cc:60 Only one MYNT EYE device, select index: 0
I/synthetic.cc:52 camera calib model: kannala_brandt
2.5.0I/utils.cc:99 MYNT EYE requests:
I/utils.cc:102   index: 0, request: width: 752, height: 480, format: Format::YUYV, fps: 25
I/utils.cc:102   index: 1, request: width: 376, height: 240, format: Format::YUYV, fps: 25
I/utils.cc:113 There are 2 stream requests, select index:

FengyuGuo, 2022-11-06:
@wenxini234 have you ever solve the problem in Ubuntu 1804? I got the problem in one of my computer but no problem in another. Quite strange problem.