slightech/MYNT-EYE-S-SDK #65: Got "/bin/bash: catkin_make : command not found" error when make ros

Zhenghong-Lai, 2019-09-19:
OS: ubuntu 16.04 LTS
ROS: ros-kinetic

I got an error when make ros :

/bin/bash: catkin_make : command not found
Makefile: 180: recipe for target ‘ros' failed
make:***[ros] Error 127

How to solve it please ?

Zhenghong-Lai, 2019-09-19:
I just open another teriminal. And execute commands below
cd ./wrapper/ros
catkin_make

This seems work well for me, though I still don't know why the Error 127 happened .

harjeb, 2019-09-25:
Hi, we have a installation video  from [here](https://www.bilibili.com/video/av68237965)

Zhenghong-Lai, 2019-09-26:
@harjeb unfortunately, the error still occurs following the video instructions

harjeb, 2019-09-26:
Hi, this system in video is made by official Ubuntu LTS 18.04.3 image. You can try in clean OS , like vms.

Zhenghong-Lai, 2019-09-26:
@harjeb Thanks a lot.Though I still don't know why the error happened, but I have to move to the next step - MYNT-EYE-VINS-Sample now. I figured it out by the method I commented above, every thing seems to work well until now

closed 2019-09-26 by Zhenghong-Lai