slightech/MYNT-EYE-S-SDK #67: How to perform calibration for model S?

alter-sachin, 2019-09-24:
Hi , I am using your SDK to collect depth maps but there seems to be a problem of calibration.
Distances that are greater than 10meters are detected as less than 6m and so on.

There is a calibration tool for model D here :
https://github.com/slightech/MYNT-EYE-D-SDK/blob/ea419d0a5a1e0ce7cca3b9e2b19eee86896e6e0c/docs/tools/calibration_tool.rst

Is there something similar for model S?
Or is there a manual way to make sure the depths come correct for me ?

Please help!

harjeb, 2019-09-25:
if you use Ubuntu 16.04, you can download calibration tool from [here](https://drive.google.com/open?id=1RsV2WEKAsfxbn-Z5nGjk5g3ml1UDEsDc)
after install, use `mynteye-s-calibrator -h` for details

alter-sachin, 2019-09-25:
Is this necessary before starting to use the camera?

harjeb, 2019-09-25:
It‘s not necessary, but if you find your depth image broken, you should calibrate device by using this tool.  And you can refer to [link](https://mynt-eye-s-sdk.readthedocs.io/en/latest/src/product/product_spec.html) for recommend depth working distance

closed 2019-11-26 by alter-sachin