slightech/MYNT-EYE-S-SDK #70: Aligning left and right IR frames with depth frame?

alter-sachin, 2019-10-10:
Hello , I have been able to use the MYNT eye S for some depth sensing work.

I want to align the left right IR with the generated depth map. Is there any way to do this ?

This is necessary for me because i want to apply some masking using the IR frames onto the depth. To reduce false positives.

Please help?

alter-sachin, 2019-10-10:
for reference please take a look at the picture.
[image]

The above picture the person is not starting on the left in the depth frame like in the IR frames.

I hope that shows the problem

harjeb, 2019-10-10:
Hi, the depth image seems not normal, you can try `get_depth_with_region` sample to check the depth distance. if it's abnormal, you need follow [this](https://mynt-eye-s-sdk.readthedocs.io/en/v2.4.1/src/tools/calibration_tool.html) to calibrate device.
Otherwise SDK have 2 disparity computing methods,  follow [this](https://mynt-eye-s-sdk.readthedocs.io/en/latest/src/sdk/data/get_disparity.html) to modify.

alter-sachin, 2019-10-10:
[image]

this is the SGBM based depth below . can you check if that looks fine to you?
i'll perform calibration agaiin

harjeb, 2019-10-10:
This depth image is broken, need calibrate. P.S. if the reprojection error value is too big, reduce -n value.