slightech/MYNT-EYE-S-SDK #74: IMU calibration

pzhu008, 2019-11-08:
Hi, 
Error occured when I use  imu_utils to calibrate IMU. (no results for gyro)  

I recorded a rosbag of IMU raw data for 90 mins. All  gyro_x, gyro_y, gyro_z are zeros. Then I use ./record to generate a dataset. Also all gyro_x, gyro_y, gyro_z are zeros. 

Do you have any idea about this?

Thanks,

niekez, 2020-01-17:
It seems that there's is some filtering being done setting the gyro values to zero when there is little movement. This makes it impossible to measure the bias and bias random walk.

It would be more appropriate to send the raw data as the topic says.

I think the firmware had to be adjusted for this, could you please make this happen?

I'm experiencing some issues  in maplab where the frame gets ignored because the gyro data is zero, also the gyro intrinsics can't be measured for callibration purposes.

pzhu008, 2020-01-17:
> It seems that there's is some filtering being done setting the gyro values to zero when there is little movement. This makes it impossible to measure the bias and bias random walk.
>
> It would be more [...]

pzhu008, 2020-01-17:
try SDK 2.3.6 without filter

niekez, 2020-01-20:
> try SDK 2.3.6 without filter

It's not possible to downgrade the firmware for 2.3.6. Also I'm using the S1030, and as far as I know there's no processing being done by the SDK. I think it's happening on a firmware level, but maybe someone can point me to the SDK code where filtering is done on the gyro.

lesurp, 2020-08-04:
Hi, any update on this issue?
I'm using the example noise values given in the repository here but I would very much prefer calibrating my mynteye myself...

lesurp, 2020-08-06:
Actually, this filtering is actively deteriorating the processing I am doing with the values. Is there any reason why the filters cannot be disabled on the s1030, while it can on the other devices? (https://slightech.github.io/MYNT-EYE-S-SDK-Guide/src/control/imu_low_pass_filter.html)

KalmanXiong, 2020-08-06:
Sorry，the company is dead..

| |
熊若阳
|
|
邮箱：ruoyangxiong@163.com
|

签名由 网易邮箱大师 定制

On 08/06/2020 17:12, Paul Lesur wrote:

Actually, this filtering is actively deteriorating the processing I am doing with the values. Is there any reason why the filters cannot be disabled on the s1030, while it can on the other devices? (https://slightech.github.io/MYNT-EYE-S-SDK-Guide/src/control/imu_low_pass_filter.html)

LSXiang, 2021-01-28:
This is a bug in SDK version 2.5, you can ref to #113