slightech/MYNT-EYE-S-SDK #77: process has died [pid 8999, exit code -11, cmd /home/nvidia/MYNT-EYE-S-SDK/wrappers/ros/devel/lib/mynt_eye_ros_wrapper/mynteye_wrapper_node __name:=mynteye_wrapper_node __log:=/home/nvidia/.ros/log/2644772c-0b64-11ea-8a93-00044bc70153/mynteye-mynteye_wrapper_node-2.log]. log file: /home/nvidia/.ros/log/2644772c-0b64

xxzzll11111, 2019-11-20:
The computer cannot find the MYNT EYE in ros node. What's wrong with it?

[mynteye/mynteye_wrapper_node-2] restarting process
process[mynteye/mynteye_wrapper_node-2]: started with pid [8999]
[ INFO] [1574233998.783553540]: Initializing nodelet with 4 worker threads.
[ INFO] [1574233998.987546923]: Detecting MYNT EYE devices
[FATAL] [1574233998.988868659]: No MYNT EYE devices :(
[ INFO] [1574233998.989043184]: MYNT EYE devices:
[ INFO] [1574233998.997456794]: Only one MYNT EYE device, select index: 0
[mynteye/mynteye_wrapper_node-2] process has died [pid 8999, exit code -11, cmd /home/nvidia/MYNT-EYE-S-SDK/wrappers/ros/devel/lib/mynt_eye_ros_wrapper/mynteye_wrapper_node __name:=mynteye_wrapper_node __log:=/home/nvidia/.ros/log/2644772c-0b64-11ea-8a93-00044bc70153/mynteye-mynteye_wrapper_node-2.log].
log file: /home/nvidia/.ros/log/2644772c-0b64-11ea-8a93-00044bc70153/mynteye-mynteye_wrapper_node-2*.log

harjeb, 2019-11-25:
.Hi, which camera model do you use?

zhangjun-xyz, 2020-06-22:
I have exactly the same problem. My device is S1030 standard.

ludi1996, 2020-10-29:
same problem. How to fix it

huanglilong, 2021-10-17:
fixed diff here: huanglilong@3f8aad8

Ubuntu20.04
OpenCV 3.4.8