slightech/MYNT-EYE-S-SDK #78: unable to compile samples on ubuntu 18.04.3

nvoelzow, 2019-11-22:
after successfully running ```make init``` building the samples fails with the following error:

```
~/MYNT-EYE-S-SDK$ make samples
Make uninstall
sudo rm -rf /usr/local/include/mynteye/
[sudo] Passwort für putz: 
sudo rm -rf /usr/local/lib/libmynteye.so*
sudo rm -rf /usr/local/lib/cmake/mynteye/
sudo rm -rf /usr/local/share/mynteye/
Make build
CD: ./_build
-- HOST_ARCH: x86_64
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:567 (message):
  Imported targets and dependency information not available for Boost version
  (all versions older than 1.33)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  cmake/Option.cmake:48 (find_package)
  CMakeLists.txt:29 (include)

-- 
-- Platform:
--   HOST_OS: Linux
--   HOST_NAME: Ubuntu
--   HOST_ARCH: x86_64
--   HOST_COMPILER: GNU
--     COMPILER_VERSION: 7.4.0
--     COMPILER_VERSION_MAJOR: 7
--     COMPILER_VERSION_MINOR: 4
--     COMPILER_VERSION_PATCH: 0
--     COMPILER_VERSION_TWEAK: 
--   OpenCV_VERSION: 4.0.0
--     OpenCV_VERSION_MAJOR: 4
--     OpenCV_VERSION_MINOR: 0
--     OpenCV_VERSION_PATCH: 0
--     OpenCV_VERSION_TWEAK: 0
--     OpenCV_VERSION_STATUS: -pre
--     OpenCV_WITH_WORLD: 
--   MYNTEYE_VERSION: 2.5.0
--     MYNTEYE_VERSION_MAJOR: 2
--     MYNTEYE_VERSION_MINOR: 5
--     MYNTEYE_VERSION_PATCH: 0
--     MYNTEYE_VERSION_TWEAK: 
-- 
-- Options:
--   WITH_API: ON
--     OpenCV: YES
--     OpenCV_VERSION: 4.0.0
--     OpenCV_WORLD: NO
--   WITH_DEVICE_INFO_REQUIRED: ON
--   WITH_CAM_MODELS: OFF
--   WITH_BOOST: ON
--     Boost: NO
--   WITH_GLOG: OFF
-- 
-- Features:
--   Filesystem: none
-- 
-- Using flag -std=c++11
-- C_FLAGS: -Wall -Wextra -march=native
-- CXX_FLAGS: -Wall -Wextra -std=c++11 -march=native
-- Namespace: mynteye
-- RPATH: /usr/local/lib
-- Configuring done
-- Generating done
-- Build files have been written to: /home/putz/MYNT-EYE-S-SDK/_build
make[1]: Verzeichnis „/home/putz/MYNT-EYE-S-SDK/_build“ wird betreten
make[2]: Verzeichnis „/home/putz/MYNT-EYE-S-SDK/_build“ wird betreten
make[3]: Verzeichnis „/home/putz/MYNT-EYE-S-SDK/_build“ wird betreten
make[3]: Verzeichnis „/home/putz/MYNT-EYE-S-SDK/_build“ wird verlassen
make[3]: Verzeichnis „/home/putz/MYNT-EYE-S-SDK/_build“ wird betreten
[  3%] Building CXX object CMakeFiles/mynteye.dir/src/mynteye/api/synthetic.cc.o
/home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc: In member function ‘void mynteye::Synthetic::InitProcessors()’:
/home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:329:41: error: ‘PointsProcessor’ was not declared in this scope
     points_processor = std::make_shared<PointsProcessor>(
                                         ^~~~~~~~~~~~~~~
/home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:329:41: note: suggested alternative: ‘points_processor’
     points_processor = std::make_shared<PointsProcessor>(
                                         ^~~~~~~~~~~~~~~
                                         points_processor
/home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:331:27: error: no matching function for call to ‘make_shared<<expression error> >(std::shared_ptr<mynteye::CameraROSMsgInfoPair>, int)’
         POINTS_PROC_PERIOD);
                           ^
In file included from /usr/include/c++/7/memory:81:0,
                 from /home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.h:19,
                 from /home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:14:
/usr/include/c++/7/bits/shared_ptr.h:703:5: note: candidate: template<class _Tp, class ... _Args> std::shared_ptr<_Tp> std::make_shared(_Args&& ...)
     make_shared(_Args&&... __args)
     ^~~~~~~~~~~
/usr/include/c++/7/bits/shared_ptr.h:703:5: note:   template argument deduction/substitution failed:
/home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:331:27: error: template argument 1 is invalid
         POINTS_PROC_PERIOD);
                           ^
/home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:336:40: error: ‘DepthProcessor’ was not declared in this scope
     depth_processor = std::make_shared<DepthProcessor>(
                                        ^~~~~~~~~~~~~~
/home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:336:40: note: suggested alternative: ‘depth_processor’
     depth_processor = std::make_shared<DepthProcessor>(
                                        ^~~~~~~~~~~~~~
                                        depth_processor
/home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:340:26: error: no matching function for call to ‘make_shared<<expression error> >(std::shared_ptr<mynteye::CameraROSMsgInfoPair>, std::shared_ptr<int>, std::shared_ptr<int>, int)’
         DEPTH_PROC_PERIOD);
                          ^
In file included from /usr/include/c++/7/memory:81:0,
                 from /home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.h:19,
                 from /home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:14:
/usr/include/c++/7/bits/shared_ptr.h:703:5: note: candidate: template<class _Tp, class ... _Args> std::shared_ptr<_Tp> std::make_shared(_Args&& ...)
     make_shared(_Args&&... __args)
     ^~~~~~~~~~~
/usr/include/c++/7/bits/shared_ptr.h:703:5: note:   template argument deduction/substitution failed:
/home/putz/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:340:26: error: template argument 1 is invalid
         DEPTH_PROC_PERIOD);
                          ^
CMakeFiles/mynteye.dir/build.make:614: recipe for target 'CMakeFiles/mynteye.dir/src/mynteye/api/synthetic.cc.o' failed
make[3]: *** [CMakeFiles/mynteye.dir/src/mynteye/api/synthetic.cc.o] Error 1
make[3]: Verzeichnis „/home/putz/MYNT-EYE-S-SDK/_build“ wird verlassen
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/mynteye.dir/all' failed
make[2]: *** [CMakeFiles/mynteye.dir/all] Error 2
make[2]: Verzeichnis „/home/putz/MYNT-EYE-S-SDK/_build“ wird verlassen
Makefile:129: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Verzeichnis „/home/putz/MYNT-EYE-S-SDK/_build“ wird verlassen
Makefile:85: recipe for target 'build' failed
make: *** [build] Error 2
```

nvoelzow, 2019-11-23:
I've been digging a bit deeper and found that the headers depth_processor.h etc are not included because ```#ifdef WITH_CAM_MODELS``` evaluates to false for whatever reason.
After commenting this out and fixing a hand full of include paths (for example in equidistant_camera.h changing "api/camera_models/camera" to just "camera.h") the compilation is now finishing without errors, but the linking fails:

```
[  1%] Linking CXX executable ../_output/bin/record
/usr/local/lib/libmynteye.so.2.5.0: undefined reference to `mynteye::PointsProcessor::PointsProcessor(std::shared_ptr<mynteye::CameraROSMsgInfoPair>, int)'
/usr/local/lib/libmynteye.so.2.5.0: undefined reference to `mynteye::DepthProcessor::DepthProcessor(std::shared_ptr<mynteye::CameraROSMsgInfoPair>, std::shared_ptr<int>, std::shared_ptr<int>, int)'
collect2: error: ld returned 1 exit status
```

nvoelzow, 2019-11-23:
ah, and the following quick patch fixes the OpenCV naming issues I also ran into:

```
diff --git a/src/mynteye/api/processor/rectify_processor.h b/src/mynteye/api/processor/rectify_processor.h
index b5449e1..4f95ebc 100644
--- a/src/mynteye/api/processor/rectify_processor.h
+++ b/src/mynteye/api/processor/rectify_processor.h
@@ -29,6 +29,9 @@
 #include <opencv2/calib3d/calib3d.hpp>
 #include <opencv2/highgui/highgui.hpp>
 
+#define CvMat cv::Mat
+#define CvSize cv::Size
+
 MYNTEYE_BEGIN_NAMESPACE
 
 class Device;
```

harjeb, 2019-11-25:
Hi this issue caused by Opencv 4.X , please install Opencv 3.X instead.

nvoelzow, 2019-11-25:
I'll give that a try - I would suggest to update the SDK to use support a recent version of OpenCV or at least update the docs to make sure that the right version of OpenCV is installed (the reference to the installation guide of OpenCV will automatically lead to installing 4.x)

harjeb, 2019-11-25:
We‘ll update the doc in next release,Thanks for giving advice

MiladGhorbaniG, 2020-01-08:
go fuck yourself

BernardinD, 2020-02-10:
Has there been any progress on this? I'm having the same issue

kkufieta, 2021-03-09:
Same issue here.