slightech/MYNT-EYE-S-SDK #79: PPA samples not working on ubuntu

nvoelzow, 2019-11-23:
after installing the sdk through the PPA as mentioned in the docs on a fresh ubuntu 18.04.3, all of the "get_*" and "camera_*" samples will crash except get_device_info and get_imu_params:

```
/opt/mynt-eye-s-sdk/samples$ ./get_device_info 
I/utils.cc:35 Detecting MYNT EYE devices
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 02831F1700090807, firmware: 2.5
I/utils.cc:60 Only one MYNT EYE device, select index: 0
E/synthetic.cc:365 Unknow calib model type in device: CalibrationModel::KANNALA_BRANDT
2.5.0I/get_device_info.cc:23 Device name: MYNT-EYE-S1030
I/get_device_info.cc:24 Serial number: 02831F1700090807
I/get_device_info.cc:25 Firmware version: 2.5
I/get_device_info.cc:26 Hardware version: 2.0
I/get_device_info.cc:27 Spec version: 1.0
I/get_device_info.cc:28 Lens type: 0000
I/get_device_info.cc:29 IMU type: 0000
I/get_device_info.cc:30 Nominal baseline: 0
I/get_device_info.cc:31 Auxiliary chip version: 0.0
I/get_device_info.cc:33 Isp version: 0
```

```
/opt/mynt-eye-s-sdk/samples$ ./camera_with_junior_device_api 
I/utils.cc:35 Detecting MYNT EYE devices
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 02831F1700090807, firmware: 2.5
I/utils.cc:60 Only one MYNT EYE device, select index: 0
I/utils.cc:99 MYNT EYE requests:
I/utils.cc:102   index: 0, request: width: 752, height: 480, format: Format::YUYV, fps: 25
I/utils.cc:102   index: 1, request: width: 376, height: 240, format: Format::YUYV, fps: 25
I/utils.cc:113 There are 2 stream requests, select index: 
0
Illegal instruction (core dumped)
```

```
/opt/mynt-eye-s-sdk/samples$ ./get_imu
I/utils.cc:35 Detecting MYNT EYE devices
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 02831F1700090807, firmware: 2.5
I/utils.cc:60 Only one MYNT EYE device, select index: 0
E/synthetic.cc:365 Unknow calib model type in device: CalibrationModel::KANNALA_BRANDT
2.5.0I/utils.cc:99 MYNT EYE requests:
I/utils.cc:102   index: 0, request: width: 752, height: 480, format: Format::YUYV, fps: 25
I/utils.cc:102   index: 1, request: width: 376, height: 240, format: Format::YUYV, fps: 25
I/utils.cc:113 There are 2 stream requests, select index: 
0
E/synthetic.cc:137 ERROR: no suited processor for stream Stream::LEFT_RECTIFIED
E/synthetic.cc:105 Unknow calib model type in device

Segmentation fault (core dumped)
```

harjeb, 2019-11-25:
Hi, Please uninstall SDK complied with source code before install PPA.

nvoelzow, 2019-11-25:
just tried starting with a fresh 18.04.3 install and then following the guide, the problem remains the same:

```
$ sudo add-apt-repository ppa:slightech/mynt-eye-s-sdk
$ sudo apt-get update
$ sudo apt-get install mynt-eye-s-sdk
$ cd /opt/mynt-eye-s-sdk/samples
$ ./camera_with_junior_device_api 
I/utils.cc:35 Detecting MYNT EYE devices
Illegal instruction (core dumped)
$ ./camera_with_senior_api 
I/utils.cc:35 Detecting MYNT EYE devices
Illegal instruction (core dumped)
```

just to check the opencv version:

```
$ apt list | grep opencv

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

cl-opencv-apps/bionic,bionic 1.11.14-1build4 all
gstreamer1.0-opencv/bionic-updates 1.14.5-0ubuntu1~18.04.1 amd64
libgstreamer-opencv1.0-0/bionic-updates 1.14.5-0ubuntu1~18.04.1 amd64
libopencv-apps-dev/bionic 1.11.14-1build4 amd64
libopencv-apps0d/bionic 1.11.14-1build4 amd64
libopencv-calib3d-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-calib3d3.2/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed,automatic]
libopencv-contrib-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-contrib3.2/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-core-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-core3.2/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed,automatic]
libopencv-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-features2d-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-features2d3.2/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed,automatic]
libopencv-flann-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-flann3.2/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed,automatic]
libopencv-highgui-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-highgui3.2/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed,automatic]
libopencv-imgcodecs-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-imgcodecs3.2/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed,automatic]
libopencv-imgproc-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-imgproc3.2/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed,automatic]
libopencv-ml-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-ml3.2/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed,automatic]
libopencv-objdetect-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-objdetect3.2/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-photo-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-photo3.2/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-shape-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-shape3.2/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-stitching-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-stitching3.2/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-superres-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-superres3.2/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-ts-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-video-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-video3.2/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-videoio-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-videoio3.2/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed,automatic]
libopencv-videostab-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-videostab3.2/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-viz-dev/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv-viz3.2/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
libopencv3.2-java/bionic-updates,bionic-updates,bionic-security,bionic-security 3.2.0+dfsg-4ubuntu0.1 all
libopencv3.2-jni/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
opencv-data/bionic-updates,bionic-updates,bionic-security,bionic-security 3.2.0+dfsg-4ubuntu0.1 all
opencv-doc/bionic-updates,bionic-updates,bionic-security,bionic-security 3.2.0+dfsg-4ubuntu0.1 all
python-opencv/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
python-opencv-apps/bionic,bionic 1.11.14-1build4 all
python3-opencv/bionic-updates,bionic-security 3.2.0+dfsg-4ubuntu0.1 amd64
ros-opencv-apps/bionic 1.11.14-1build4 amd64
```

harjeb, 2019-11-25:
Hi, we have checked on fresh 18.04.3 too, seems like no problem, you could try with sudo command. [image]

nvoelzow, 2019-11-25:
same thing with sudo, all sample applications crash with the same error:

```
$ sudo ./get_device_info 
I/utils.cc:35 Detecting MYNT EYE devices
Illegal instruction
```

and when running in gdb:
```
(gdb) run
Starting program: /opt/mynt-eye-s-sdk/samples/get_imu_params 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
I/utils.cc:35 Detecting MYNT EYE devices

Program received signal SIGILL, Illegal instruction.
0x00007ffff7af53ae in mynteye::uvc::query_devices(std::shared_ptr<mynteye::uvc::context>) () from /usr/lib/libmynteye.so.2
(gdb) bt
#0  0x00007ffff7af53ae in mynteye::uvc::query_devices(std::shared_ptr<mynteye::uvc::context>) () from /usr/lib/libmynteye.so.2
#1  0x00007ffff7b23226 in mynteye::Context::Context() () from /usr/lib/libmynteye.so.2
#2  0x00007ffff7b54d7f in mynteye::device::select() () from /usr/lib/libmynteye.so.2
#3  0x00007ffff7b61320 in mynteye::API::Create(int, char**) () from /usr/lib/libmynteye.so.2
#4  0x000055555555598d in main ()
```

The camera is showing up in lsusb as two uvcvideo devices (creating /dev/video0 and /dev/video1):
```
$ lsusb -t
/:  Bus 10.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 5000M
/:  Bus 09.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 5000M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 5000M
    |__ Port 1: Dev 3, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 3, If 1, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 3: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 7: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 8: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 8: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
```

nvoelzow, 2019-11-25:
opening /dev/video0 in xawtv or vlc works by the way, giving me a green-purple overlay image of the two camera views.

harjeb, 2019-11-26:
Hi could you try `ls /usr/local/lib | grep mynt`

fukamike, 2019-12-02:
I have same problem.
I tried ls /usr/local/lib | grep mynt
but no massage.
[image]

fukamike, 2019-12-02:
I using ubuntu 18.04.3 .
I tried samples.
`$ cd /opt/mynt-eye-s-sdk/samples
$ ./camera_with_junior_device_api `
the result is `I/utils.cc:35 Detecting MYNT EYE devices
Illegal instruction (core dumped)`
[image]

harjeb, 2019-12-02:
@fukamike Hi, which model do you use？ `ls /usr/local/lib | grep mynt` is used to check .so file compiled by source code.

fukamike, 2019-12-02:
I'm using EYE s1030

harjeb, 2019-12-02:
@fukamike  Could you use source code to compile?  we don't meet this issue on our environment(18.04.3 64-bit PC with USB3.0 interface).

fukamike, 2019-12-03:
@harjeb  I was able to compile source code and run sample code.
[image]

fukamike, 2019-12-03:
If I want to develop codes, Should I change sample codes?
For example get_stereo_rectified.cc ..etc.

harjeb, 2019-12-03:
Yes, you can change sample codes or follow the demo project to create a new one.

simonegodio1994, 2022-01-25:
@fukamike Excuse me I have the same problem.. did you succeed to make it works also the IMU? Can you please share it?

Thank you in advance!