slightech/MYNT-EYE-S-SDK #85: it seemes some of the file is wrong

61269240, 2020-01-01:
I install it with the comment:
make install

And then when the process get 72%,I see the following error:
[ 72%] Building CXX object CMakeFiles/mynteye.dir/src/mynteye/api/synthetic.cc.o
/home/pi/Documents/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc: In member function ‘void mynteye::Synthetic::InitProcessors()’:
/home/pi/Documents/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc:329:41: error: ‘PointsProcessor’ was not declared in this scope
     points_processor = std::make_shared<PointsProcessor>(
                                         ^~~~~~~~~~~~~~~
/home/pi/Documents/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc:331:27: error: no matching function for call to ‘make_shared(std::shared_ptr<mynteye::CameraROSMsgInfoPair>, int)’
         POINTS_PROC_PERIOD);
                           ^
In file included from /usr/include/c++/6/memory:82:0,
                 from /home/pi/Documents/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.h:19,
                 from /home/pi/Documents/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc:14:
/usr/include/c++/6/bits/shared_ptr.h:632:5: note: candidate: template<class _Tp, class ... _Args> std::shared_ptr<_Tp1> std::make_shared(_Args&& ...)
     make_shared(_Args&&... __args)
     ^~~~~~~~~~~
/usr/include/c++/6/bits/shared_ptr.h:632:5: note:   template argument deduction/substitution failed:
/home/pi/Documents/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc:331:27: error: template argument 1 is invalid
         POINTS_PROC_PERIOD);
                           ^
/home/pi/Documents/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc:336:40: error: ‘DepthProcessor’ was not declared in this scope
     depth_processor = std::make_shared<DepthProcessor>(
                                        ^~~~~~~~~~~~~~
/home/pi/Documents/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc:340:26: error: no matching function for call to ‘make_shared(std::shared_ptr<mynteye::CameraROSMsgInfoPair>, std::shared_ptr<int>, std::shared_ptr<int>, int)’
         DEPTH_PROC_PERIOD);
                          ^
In file included from /usr/include/c++/6/memory:82:0,
                 from /home/pi/Documents/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.h:19,
                 from /home/pi/Documents/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc:14:
/usr/include/c++/6/bits/shared_ptr.h:632:5: note: candidate: template<class _Tp, class ... _Args> std::shared_ptr<_Tp1> std::make_shared(_Args&& ...)
     make_shared(_Args&&... __args)
     ^~~~~~~~~~~
/usr/include/c++/6/bits/shared_ptr.h:632:5: note:   template argument deduction/substitution failed:
/home/pi/Documents/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc:340:26: error: template argument 1 is invalid
         DEPTH_PROC_PERIOD);
                          ^
CMakeFiles/mynteye.dir/build.make:361: recipe for target 'CMakeFiles/mynteye.dir/src/mynteye/api/synthetic.cc.o' failed
make[3]: *** [CMakeFiles/mynteye.dir/src/mynteye/api/synthetic.cc.o] Error 1
make[3]: Leaving directory '/home/pi/Documents/MYNT-EYE-S-SDK-master/_build'
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/mynteye.dir/all' failed
make[2]: *** [CMakeFiles/mynteye.dir/all] Error 2
make[2]: Leaving directory '/home/pi/Documents/MYNT-EYE-S-SDK-master/_build'
Makefile:129: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/pi/Documents/MYNT-EYE-S-SDK-master/_build'
Makefile:84: recipe for target 'build' failed
make: *** [build] Error 2

maybe there is some problem about your code

harjeb, 2020-01-02:
Don't support raspberry pi cause to performance problem

61269240, 2020-01-25:
But I think build error is not a performance problem

miksarus, 2021-06-09:
Same problem on Linux Mint 20.1
`/home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc: In member function ‘void mynteye::Synthetic::InitProcessors()’:
/home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:329:41: error: ‘PointsProcessor’ was not declared in this scope; did you mean ‘points_processor’?
  329 |     points_processor = std::make_shared<PointsProcessor>(
      |                                         ^~~~~~~~~~~~~~~
      |                                         points_processor
/home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:331:27: error: no matching function for call to ‘make_shared<<expression error> >(std::shared_ptr<mynteye::CameraROSMsgInfoPair>, int)’
  331 |         POINTS_PROC_PERIOD);
      |                           ^
In file included from /usr/include/c++/9/memory:81,
                 from /home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.h:19,
                 from /home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:14:
/usr/include/c++/9/bits/shared_ptr.h:714:5: note: candidate: ‘template<class _Tp, class ... _Args> std::shared_ptr<_Tp> std::make_shared(_Args&& ...)’
  714 |     make_shared(_Args&&... __args)
      |     ^~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:714:5: note:   template argument deduction/substitution failed:
/home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:331:27: error: template argument 1 is invalid
  331 |         POINTS_PROC_PERIOD);
      |                           ^
/home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:336:40: error: ‘DepthProcessor’ was not declared in this scope; did you mean ‘depth_processor’?
  336 |     depth_processor = std::make_shared<DepthProcessor>(
      |                                        ^~~~~~~~~~~~~~
      |                                        depth_processor
/home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:340:26: error: no matching function for call to ‘make_shared<<expression error> >(std::shared_ptr<mynteye::CameraROSMsgInfoPair>, std::shared_ptr<int>, std::shared_ptr<int>, int)’
  340 |         DEPTH_PROC_PERIOD);
      |                          ^
In file included from /usr/include/c++/9/memory:81,
                 from /home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.h:19,
                 from /home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:14:
/usr/include/c++/9/bits/shared_ptr.h:714:5: note: candidate: ‘template<class _Tp, class ... _Args> std::shared_ptr<_Tp> std::make_shared(_Args&& ...)’
  714 |     make_shared(_Args&&... __args)
      |     ^~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:714:5: note:   template argument deduction/substitution failed:
/home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:340:26: error: template argument 1 is invalid
  340 |         DEPTH_PROC_PERIOD);
      |                          ^
`

miksarus, 2021-06-09:
I looked at the code and find, that errors is becouse `WITH_CAM_MODELS` flag is unset. But if i set it manualy build fails with other error:
```
[ 72%] Building CXX object CMakeFiles/mynteye.dir/src/mynteye/api/synthetic.cc.o
In file included from /home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.h:27,
                 from /home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/processor/depth_processor.h:22,
                 from /home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:31:
/home/miksarus/opt/MYNT-EYE-S-SDK/src/mynteye/api/camera_models/equidistant_camera.h:21:10: fatal error: api/camera_models/camera.h: No such file or directory
   21 | #include "api/camera_models/camera.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/mynteye.dir/build.make:362: CMakeFiles/mynteye.dir/src/mynteye/api/synthetic.cc.o] Error 1
make[3]: Leaving directory '/home/miksarus/opt/MYNT-EYE-S-SDK/_build'
make[2]: *** [CMakeFiles/Makefile2:76: CMakeFiles/mynteye.dir/all] Error 2
make[2]: Leaving directory '/home/miksarus/opt/MYNT-EYE-S-SDK/_build'
make[1]: *** [Makefile:130: all] Error 2
make[1]: Leaving directory '/home/miksarus/opt/MYNT-EYE-S-SDK/_build'
make: *** [Makefile:85: build] Error 2
```

djoshuadulle, 2021-09-03:
I'm not saying this is a good fix, but I found that in `MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc` a small rearrangement moving the points_processor and depth_processor from lines 329 and 336 fixed the build.

Here's the new code section:

`  if (calib_model_ == CalibrationModel::PINHOLE) {
    // PINHOLE
    auto &&rectify_processor_ocv =
        std::make_shared<RectifyProcessorOCV>(intr_left_, intr_right_, extr_,
                                              RECTIFY_PROC_PERIOD);
    rectify_processor = rectify_processor_ocv;

    auto disparity_processor_imp =
      std::make_shared<DisparityProcessor>(DisparityComputingMethod::BM,
                                           nullptr,
                                           DISPARITY_PROC_PERIOD);
    disparity_processor = disparity_processor_imp;
#ifdef WITH_CAM_MODELS
    points_processor = std::make_shared<PointsProcessor>(
        rectify_processor_ocv -> getCameraROSMsgInfoPair(),
        POINTS_PROC_PERIOD);
    depth_processor = std::make_shared<DepthProcessor>(
        rectify_processor_ocv -> getCameraROSMsgInfoPair(),
        disparity_processor_imp->GetMinDisparity(),
        disparity_processor_imp->GetMaxDisparity(),
        DEPTH_PROC_PERIOD);
  } else if (calib_model_ == CalibrationModel::KANNALA_BRANDT) {
    // KANNALA_BRANDT
    auto rectify_processor_imp =
        std::make_shared<RectifyProcessor>(intr_left_, intr_right_, extr_,
                                           RECTIFY_PROC_PERIOD);
    rectify_processor = rectify_processor_imp;
    points_processor = std::make_shared<PointsProcessor>(
        rectify_processor_imp -> getCameraROSMsgInfoPair(),
        POINTS_PROC_PERIOD);
    auto disparity_processor_imp =
      std::make_shared<DisparityProcessor>(DisparityComputingMethod::BM,
          rectify_processor_imp -> getCameraROSMsgInfoPair(),
          DISPARITY_PROC_PERIOD);
    depth_processor = std::make_shared<DepthProcessor>(
        rectify_processor_imp -> getCameraROSMsgInfoPair(),
        disparity_processor_imp->GetMinDisparity(),
        disparity_processor_imp->GetMaxDisparity(),
        DEPTH_PROC_PERIOD);
    disparity_processor = disparity_processor_imp;
#endif
  } else {
    // UNKNOW`