slightech/MYNT-EYE-S-SDK #86: How to read RGB frames using cv2.VideoCapture

arashmob, 2020-01-08:
hello everyone. I'm currently using cv2.VideoCapture in python to read my mynt s1030 frames. however, the problem is that it outputs grayscale images. is there any way to read RGB left and right frames? (i don't want to use sdk because i need frames in python) 

I appreciate any comment. thanks

buchi77, 2020-01-09:
hello arashmob.
my simple test code. 
```python
import numpy as np
import cv2

def main():
    cam_id = 0 #select camera ID

    width = 752
    height = 480

    cam = cv2.VideoCapture(cam_id)
    
    cam.set(cv2.CAP_PROP_FRAME_WIDTH,width)
    cam.set(cv2.CAP_PROP_FRAME_HEIGHT,height)
    cam.set(cv2.CAP_PROP_CONVERT_RGB,0) # 0 = output raw_data, 1 = RGB

    left_data = np.empty((width * height),int)
    right_data = np.empty((width * height),int)

    while True:

        ret,frame = cam.read()

        left_data = frame[0,::2]
        right_data = frame[0,1::2]

        cv2.imshow("left",np.reshape(left_data,(height,width)))
        cv2.imshow("right",np.reshape(right_data,(height,width)))
            
        key = cv2.waitKey(1)
        if key == ord('q'):
            break

    cam.release()
    cv2.destroyAllWindows()

if __name__== '__main__':
    main()
```

arashmob, 2020-01-09:
thanks for the comment. I tried your code with setting cv2.CAP_PROP_CONVERT_RGB to 1, but it ran into problem. are you able to read RGB using your code?

arashmob, 2020-01-09:
> hello arashmob.
> my simple test code.
>
> ```python
> import numpy as np
> import cv2
>
> def main():
>     cam_id = 0 #select camera ID
>
>     width = 752
>     height = 480
>
>     cam = [...]

thanks for the comment. I tried your code with setting cv2.CAP_PROP_CONVERT_RGB to 1, but it ran into problem. are you able to read RGB using your code?

buchi77, 2020-01-09:
this code is raw_data only. don't use 1.

s1030 type is monochrome only...

arashmob, 2020-01-10:
> this code is raw_data only. don't use 1.
> 
> s1030 type is monochrome only...

Oh thanks

closed 2020-01-11 by arashmob