slightech/MYNT-EYE-S-SDK #88: problem with rectification in python

arashmob, 2020-01-15:
Hello Everyone,
I want my raw images (I read them in python) to be rectified. but what i need is the exact intrinsic and extrinsic s1030 parameters. I'm using the parameters that its file is placed in /samples/config/S1030/imag.params.pinhole 

I don't know they are correct or not.
I dont want using SDK anymore so what i want is the exact params and how to rectify raw images in python. 

I appreciate any comments

MiladGhorbaniG, 2020-01-17:
Hi arashmob, thanks for question. Since the cameras are fisheye type, you probably should rectify the images using kannala brandt model.

arashmob, 2020-01-17:
> Hi arashmob, thanks for question. Since the cameras are fisheye type, you probably should rectify the images using kannala brandt model instead of pinhole model.

Thanks Milad. your help is greatly appreciated

closed 2020-01-17 by arashmob