slightech/MYNT-EYE-S-SDK #92: compile error caused by opencv version 3.4.9

gogogoKKK, 2020-02-17:
Current platform to compile the lastest SDK:
```
-- HOST_ARCH: x86_64
-- 
-- Platform:
--   HOST_OS: Linux
--   HOST_NAME: Ubuntu
--   HOST_ARCH: x86_64
--   HOST_COMPILER: GNU
--     COMPILER_VERSION: 7.4.0
--     COMPILER_VERSION_MAJOR: 7
--     COMPILER_VERSION_MINOR: 4
--     COMPILER_VERSION_PATCH: 0
--     COMPILER_VERSION_TWEAK: 
--   OpenCV_VERSION: 3.4.9
--     OpenCV_VERSION_MAJOR: 3
--     OpenCV_VERSION_MINOR: 4
--     OpenCV_VERSION_PATCH: 9
--     OpenCV_VERSION_TWEAK: 0
--     OpenCV_VERSION_STATUS: 
--     OpenCV_WITH_WORLD: 
--   MYNTEYE_VERSION: 2.5.0
--     MYNTEYE_VERSION_MAJOR: 2
--     MYNTEYE_VERSION_MINOR: 5
--     MYNTEYE_VERSION_PATCH: 0
--     MYNTEYE_VERSION_TWEAK: 
-- 
-- Options:
--   WITH_API: ON
--     OpenCV: YES
--     OpenCV_VERSION: 3.4.9
--     OpenCV_WORLD: NO
--   WITH_DEVICE_INFO_REQUIRED: ON
--   WITH_CAM_MODELS: ON
--   WITH_BOOST: ON
--     Boost: YES
--     Boost_VERSION: 1.65.1
--   WITH_GLOG: OFF
-- 
-- Features:
--   Filesystem: boost
```

The error is given as
```
In file included from /home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:14:0:
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.h:93:31: error: could not convert ‘cv::Size_<int>()’ from ‘cv::Size {aka cv::Size_<int>}’ to ‘CvSize’
       CvSize newImgSize = cv::Size());
                               ^~~~~~
In file included from /home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/depth_processor.h:22:0,
                 from /home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/depth_processor.cc:14:
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.h:93:31: error: could not convert ‘cv::Size_<int>()’ from ‘cv::Size {aka cv::Size_<int>}’ to ‘CvSize’
       CvSize newImgSize = cv::Size());
                               ^~~~~~
In file included from /home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/depth_processor.h:22:0,
                 from /home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:31:
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.h:93:31: error: could not convert ‘cv::Size_<int>()’ from ‘cv::Size {aka cv::Size_<int>}’ to ‘CvSize’
       CvSize newImgSize = cv::Size());
                               ^~~~~~
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc: In member function ‘std::shared_ptr<mynteye::CameraROSMsgInfoPair> mynteye::RectifyProcessor::stereoRectify(mynteye::models::CameraPtr, mynteye::models::CameraPtr, mynteye::IntrinsicsEquidistant, mynteye::IntrinsicsEquidistant, mynteye::Extrinsics)’:
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:329:15: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
   CvMat c_R = cv_R, c_t = cv_t;
               ^~~~
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:329:27: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
   CvMat c_R = cv_R, c_t = cv_t;
                           ^~~~
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:330:16: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
   CvMat c_K1 = K1, c_K2 = K2, c_D1 = D1, c_D2 = D2;
                ^~
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:330:27: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
   CvMat c_K1 = K1, c_K2 = K2, c_D1 = D1, c_D2 = D2;
                           ^~
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:330:38: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
   CvMat c_K1 = K1, c_K2 = K2, c_D1 = D1, c_D2 = D2;
                                      ^~
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:330:49: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
   CvMat c_K1 = K1, c_K2 = K2, c_D1 = D1, c_D2 = D2;
                                                 ^~
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:331:16: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
   CvMat c_R1 = R1, c_R2 = R2, c_P1 = P1, c_P2 = P2;
                ^~
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:331:27: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
   CvMat c_R1 = R1, c_R2 = R2, c_P1 = P1, c_P2 = P2;
                           ^~
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:331:38: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
   CvMat c_R1 = R1, c_R2 = R2, c_P1 = P1, c_P2 = P2;
                                      ^~
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:331:49: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
   CvMat c_R1 = R1, c_R2 = R2, c_P1 = P1, c_P2 = P2;
                                                 ^~
/home/zsk/Downloads/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.cc:336:19: error: no matching function for call to ‘mynteye::RectifyProcessor::stereoRectify(mynteye::models::CameraPtr&, mynteye::models::CameraPtr&, CvMat*, CvMat*, CvMat*, CvMat*, cv::Size&, CvMat*, CvMat*, CvMat*, CvMat*, CvMat*, CvMat*, double*, double*)’
       &cx1_min_cx2);
```

Through the [docs](https://docs.opencv.org/3.4.9/d3/db2/structCvMat.html) of opencv 3.4.9, the structure `CvMat` is obsolete and not recommended. You might try to upgrade the SDK to support the version 3.4.9.

cz-cz, 2020-06-21:
i have the same error 
how can i fix it?

cz-cz, 2020-06-21:
my OpenCV_VERSION: 3.4.10

cz-cz, 2020-06-22:
i built another opencv version 3.4.5 ,it works,but the  opencv  compile process is a little complex . still hope for upgrading the SDK

glabmoris, 2020-08-25:
Getting the same error, they changed the CvSize type in 3.4.9.  It's a relatively simple fix.  I'll see if I have time to patch the SDK

ludi1996, 2020-10-28:
I met the same error. Compiling with OpenCv 3.4.1
How to fix it without reinstall OpenCV

~/Desktop/MYNT-EYE-S-SDK/src/mynteye/api/processor/rectify_processor.h:90:31: error: could not convert ‘cv::Size_<int>()’ from ‘cv::Size’ {aka ‘cv::Size_<int>’} to ‘CvSize’
   90 |       CvSize newImgSize = cv::Size());
      |                               ^~~~~~
      |                               |
      |                               cv::Size {aka cv::Size_<int>}

glabmoris, 2020-11-03:
@ludi1996 You just need to update the code to use the newer OpenCV constructs (cv::Size, etc). It's actually not that big of a task to just update it. Making it compatible with all opencv versions is though, I think that's why they left it like this.

Blossom-ASL, 2021-11-25:
@glabmoris 
Can you share the updated code which use the use the newer OpenCV constructs?