slightech/MYNT-EYE-S-SDK #94: 在S1040上运行ROS Wrapper报错

luffycheung, 2020-04-22:
执行roslaunch mynt_eye_ros_wrapper mynteye.launch后报错如下
[mynteye/mynteye_wrapper_node-2] process has died [pid 1287, exit code -7, cmd /home/pi/Downloads/MYNT-EYE-S-SDK-master/wrappers/ros/devel/lib/mynt_eye_ros_wrapper/mynteye_wrapper_node __name:=mynteye_wrapper_node __log:=/home/pi/.ros/log/94e6e30e-847c-11ea-8375-dca6325f8ce7/mynteye-mynteye_wrapper_node-2.log].
log file: /home/pi/.ros/log/94e6e30e-847c-11ea-8375-dca6325f8ce7/mynteye-mynteye_wrapper_node-2*.log

closed 2020-06-03 by luffycheung

zhangjun-xyz, 2020-06-22:
How do you solve your problem?

Z-Jeff, 2022-01-19:
出现了同样的问题，请问解决了吗

> 环境：树莓派4、opencv3.4.1、ros melodic； samples sdk运行正常。 执行roslaunch mynt_eye_ros_wrapper mynteye.launch后报错如下 [mynteye/mynteye_wrapper_node-2] process has died [pid 1287, exit code -7, cmd /home/pi/Downloads/MYNT-EYE-S-SDK-master/wrappers/ros/devel/lib/mynt_eye_ros_wrapper/mynteye_wrapper_node __name:=mynteye_wrapper_node __log:=/home/pi/.ros/log/94e6e30e-847c-11ea-8375-dca6325f8ce7/mynteye-mynteye_wrapper_node-2.log]. log file: /home/pi/.ros/log/94e6e30e-847c-11ea-8375-dca6325f8ce7/mynteye-mynteye_wrapper_node-2*.log

flashYB426, 2023-02-25:
有人有解决方法吗？

Rosa712, 2024-04-01:
遇到同样的问题