slightech/MYNT-EYE-S-SDK #102: Bad quality pointcloud

songshan0321, 2020-09-22:
Device: MYNT-EYE-S1040

PointCloud quality is very bad, it seems like not even working properly. Any idea what error is this? How to solve it?

This is the output I get from running using ros wrapper in this SDK.

disparity_computing_method: 1 (bm)
[image]

disparity_computing_method: 0 (sgbm)
[image]

tiagonovo0, 2020-11-02:
Try to use [mynteye-s-calibrator](https://github.com/slightech/MYNT-EYE-S-SDK/blob/master/docs/src/tools/calibration_tool.rst). It only works with Ubuntu 16, but it was the only way I found to correctly calibrate my S1030.

Andrew-rw, 2020-11-05:
Hello, 
I have a similar issue as @songshan0321 , but when I try to run calibration I have the following error:
```
root@a646dcd58242:/opt/mynt-eye-s-sdk/samples# mynteye-s-calibrator -x 11 -y 7 -s 0.036
I/main.cc:68 Calibrate MYNT EYE S device with chessboard 11x7, size: 0.036m, num: 11
I/utils.cc:35 Detecting MYNT EYE devices
F/channels.cc:477 Check failed: n == file_size The firmware not support getting device info, you could upgrade to latest
Aborted (core dumped)
```
Below is a device info:
```
root@a646dcd58242:/opt/mynt-eye-s-sdk/samples# ./get_device_info 
I/utils.cc:35 Detecting MYNT EYE devices
E/device.cc:670 Image params not found, but we need it to process the images. Please `make tools` and use `img_params_writer` to write the image params. If you update the SDK from 1.x, the `SN*.conf` is the file contains them. Besides, you could also calibrate them by yourself. Read the guide doc (https://github.com/slightech/MYNT-EYE-SDK-2-Guide) to learn more.
W/device.cc:677 Intrinsics & extrinsics not exist. Use default intrinsics.
I/device.cc:678 camera calib model: unknow, use default pinhole data
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1040, sn: xxMANUALLY_EDITEDxxx, firmware: 0.0
I/utils.cc:60 Only one MYNT EYE device, select index: 0
I/synthetic.cc:49 camera calib model: pinhole
E/disparity_processor.cc:57 BM not supported in opencv 2.x, use sgbm
2.5.0W/version_checker.cc:158 You're using a custom mynteye device
I/get_device_info.cc:23 Device name: MYNT-EYE-S1040
I/get_device_info.cc:24 Serial number: xxMANUALLY_EDITEDxxx
I/get_device_info.cc:25 Firmware version: 0.0
I/get_device_info.cc:26 Hardware version: 3.0
I/get_device_info.cc:27 Spec version: 1.3
I/get_device_info.cc:28 Lens type: 0000
I/get_device_info.cc:29 IMU type: 0000
I/get_device_info.cc:30 Nominal baseline: 0
I/get_device_info.cc:31 Auxiliary chip version: 0.0
I/get_device_info.cc:33 Isp version: 0
```
@tiagonovo0 could you tell me do I need to update the firmware?
Thank you

tiagonovo0, 2020-11-05:
Hi, I had the same error when running on Ubuntu 16 inside docker. I know that virtual machines are not recommended too because of the usb 3.0. 
For me, it only worked when I installed ubuntu 16 from ground. 

I do not know the S1040 model but from what I see in your device info, it is look like you do not have any parameter defined so you trully need to calibrate it. 
Try to update the firmware and calibrate it.

Bests

Andrew-rw, 2020-11-05:
Hi @tiagonovo0 ,
Thank you for the quick reply. I've updated firmware (the latest what I've found was 2.7) by using Windows application. After firmware upgrade camera is recognizied as s1030: 
```
user@DEV-PC:~/MYNT-EYE-S-SDK/samples/_output/bin$ ./get_device_info 
I/utils.cc:35 Detecting MYNT EYE devices
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: xxMANUALLY_EDITEDxxx, firmware: 2.7
I/utils.cc:60 Only one MYNT EYE device, select index: 0
I/synthetic.cc:49 camera calib model: pinhole
2.5.0I/get_device_info.cc:23 Device name: MYNT-EYE-S1030
I/get_device_info.cc:24 Serial number: xxMANUALLY_EDITEDxxx
I/get_device_info.cc:25 Firmware version: 2.7
I/get_device_info.cc:26 Hardware version: 0.0
I/get_device_info.cc:27 Spec version: 1.3
I/get_device_info.cc:28 Lens type: 0000
I/get_device_info.cc:29 IMU type: 0000
I/get_device_info.cc:30 Nominal baseline: 0
I/get_device_info.cc:31 Auxiliary chip version: 0.0
I/get_device_info.cc:33 Isp version: 0

```
I also tried to set default configurations for image and imu using files provided in SDK.
But the problem is that after firmware update I can get mono and diparity images neither in ROS wrapper nor using SDK samples. Running ROS node does not provide any error except TF error:
```
user@DEV-PC:~$ roslaunch mynt_eye_ros_wrapper display.launch 
... logging to /home/user/.ros/log/e6a18944-1fa2-11eb-a463-346895594943/roslaunch-DEV-PC-3051.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://127.0.0.1:35515/

SUMMARY
========

PARAMETERS
 * /mynteye/depth/image_raw/disable_pub_plugins: ['image_transport...
 * /mynteye/disparity/image_norm/disable_pub_plugins: ['image_transport...
 * /mynteye/disparity/image_raw/disable_pub_plugins: ['image_transport...
 * /mynteye/left/image_mono/disable_pub_plugins: ['image_transport...
 * /mynteye/left/image_raw/disable_pub_plugins: ['image_transport...
 * /mynteye/left_rect/image_rect/disable_pub_plugins: ['image_transport...
 * /mynteye/left_rect_mono/disable_pub_plugins: ['image_transport...
 * /mynteye/mynteye_wrapper_node/base_frame_id: mynteye_link
 * /mynteye/mynteye_wrapper_node/depth_frame_id: mynteye_depth_frame
 * /mynteye/mynteye_wrapper_node/depth_topic: depth/image_raw
 * /mynteye/mynteye_wrapper_node/depth_type: 0
 * /mynteye/mynteye_wrapper_node/disparity_computing_method: 1
 * /mynteye/mynteye_wrapper_node/disparity_frame_id: mynteye_disparity...
 * /mynteye/mynteye_wrapper_node/disparity_norm_frame_id: mynteye_disparity...
 * /mynteye/mynteye_wrapper_node/disparity_norm_topic: disparity/image_norm
 * /mynteye/mynteye_wrapper_node/disparity_topic: disparity/image_raw
 * /mynteye/mynteye_wrapper_node/enable_depth: False
 * /mynteye/mynteye_wrapper_node/enable_disparity: False
 * /mynteye/mynteye_wrapper_node/enable_disparity_norm: False
 * /mynteye/mynteye_wrapper_node/enable_left_rect: False
 * /mynteye/mynteye_wrapper_node/enable_points: False
 * /mynteye/mynteye_wrapper_node/enable_right_rect: False
 * /mynteye/mynteye_wrapper_node/gravity: 9.8
 * /mynteye/mynteye_wrapper_node/imu_topic: imu/data_raw
 * /mynteye/mynteye_wrapper_node/is_multiple: False
 * /mynteye/mynteye_wrapper_node/left_frame_id: mynteye_left_frame
 * /mynteye/mynteye_wrapper_node/left_mono_topic: left/image_mono
 * /mynteye/mynteye_wrapper_node/left_rect_frame_id: mynteye_left_rect...
 * /mynteye/mynteye_wrapper_node/left_rect_topic: left_rect/image_rect
 * /mynteye/mynteye_wrapper_node/left_topic: left/image_raw
 * /mynteye/mynteye_wrapper_node/model_positioin_x: 0.0
 * /mynteye/mynteye_wrapper_node/model_positioin_y: -0.176
 * /mynteye/mynteye_wrapper_node/model_positioin_z: 0.0
 * /mynteye/mynteye_wrapper_node/model_rotation_x: 1.5708
[... 119 lines cut ...]
[ERROR] [1604607010.252130082]: Ignoring transform for child_frame_id "camera_imu_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 -0.000000 0.500000)

```
And running, for example, **get_disparity** sample returns "Timeout waiting for key frames" error:
```
user@DEV-PC:~/MYNT-EYE-S-SDK/samples/_output/bin$ ./get_disparity 
I/utils.cc:35 Detecting MYNT EYE devices
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: xxMANUALLY_EDITEDxxx, firmware: 2.7
I/utils.cc:60 Only one MYNT EYE device, select index: 0
I/synthetic.cc:49 camera calib model: pinhole
2.5.0I/utils.cc:99 MYNT EYE requests:
I/utils.cc:102   index: 0, request: width: 752, height: 480, format: Format::YUYV, fps: 25
I/utils.cc:102   index: 1, request: width: 376, height: 240, format: Format::YUYV, fps: 25
I/utils.cc:113 There are 2 stream requests, select index: 
0
disparity fps:
F/streams.cc:165 Timeout waiting for key frames. Please use USB 3.0, and not in virtual machine.
Aborted

```
I'm using USB 3.0 and it worked on previous unknown version of firmware. But still I **can get stereo image** using Windows application or by getting video stream directly from device (/dev/videoX)
@tiagonovo0 , maybe you faced with similar issue before?

Thank you

tiagonovo0, 2020-11-06:
Actually, I did not have those issues, even after I update the firmware to 2.7. 

Have you tried to calibrate the camera by manually? Because I think they already are calibrated with the equidistant model, Kannala Brandt. The other difference I have is `OpenGl version: 4.6 (GLSL 4.6)`.

For the command `roslaunch mynt_eye_ros_wrapper display.launch`, try to check the TFs on ROS.

Note that your hardware version has changed. That did not happen to me either. Mine is 2.0.

yashart, 2020-11-20:
@Andrew-rw , I have had a similar issue with mynteye s1040 and firmware 2.7. It is not a good fix, but I commented out checks on header and checksum for image package from camera, and it is working. For me, header is allways 0 (not 0x3B) and checksum is bad. I think, it is a firmware problem.
[fix-s1040.patch.txt](https://github.com/slightech/MYNT-EYE-S-SDK/files/5574154/fix-s1040.patch.txt)

randohinn, 2021-01-06:
Seeing this as well, sdk fix helped get a stream running again, but the pointcloud looks awful. mynteye-s-calibrator on 16.04 can't talk to the wiped hardware version firmware, and the ros camera calibrator assertion errors on calibrating..

liu83, 2022-04-13:
@Andrew-rw @yashart 
please refer to #116 where the issue can be solved by updating the firmware.

Andrew-rw, 2022-06-18:
@liu83 may I ask you to share the firmware image? Unfortunately I have difficulties downloading it from the resource which @gh18037581263 has mentioned:( 
Maybe put it as a pull request to the https://github.com/Andrew-rw/MYNT_EYE_TESTS ?

Thank you!