slightech/MYNT-EYE-S-SDK #104: make ros ：/usr/bin/ld: cannot find -lBoost::filesystem

uglymie, 2020-10-22:
make ros时出现以下错误：
/usr/bin/ld: cannot find -lBoost::filesystem
collect2: error: ld returned 1 exit status
不知道是boost库哪里出了问题，求帮忙。

xinankejidaxueliangchao, 2020-12-20:
请问你这个问题解决了吗？？

uglymie, 2020-12-21:
> 请问你这个问题解决了吗？？
没有解决，后来没有用他们家的相机了。

xinankejidaxueliangchao, 2020-12-21:
好吧，不知道是不是因为我用的是虚拟机，我在我同学他哪儿试了一下，是ok的。他不是虚拟机ubuntu，直接电脑装的是ubuntu系统，版本是18.04

25u62010, 2022-01-02:
CMakeLists里加上这两句：
find_package(Boost REQUIRED)
find_package(Boost COMPONENTS filesystem thread REQUIRED)

Karen-Cyber, 2022-01-09:
> CMakeLists里加上这两句： find_package(Boost REQUIRED) find_package(Boost COMPONENTS filesystem thread REQUIRED)

yes, it actually worked!

Z-Jeff, 2022-01-19:
我也遇到了这个问题。
我的电脑系统是Ubuntu20.04，非虚拟机。
尝试加上：
find_package(Boost REQUIRED)
find_package(Boost COMPONENTS filesystem thread REQUIRED)
也没解决这个问题。
请问有其他解决办法吗？

Z-Jeff, 2022-01-19:
> 
我搞清楚了，是在“MYNT-EYE-S-SDK/wrappers/ros/src/mynt_eye_ros_wrapper/CMakeLists.txt”中添加这两行：
find_package(Boost REQUIRED)
find_package(Boost COMPONENTS filesystem thread REQUIRED)
问题就解决了。
之前修改的是“MYNT-EYE-S-SDK/CMakeLists.txt”，无法解决问题。