slightech/MYNT-EYE-S-SDK #107: [ROS] Depth point cloud

tiagonovo0, 2020-11-02:
Hi, I have 3 mynteye S1030 and I am trying to use these to build a 3D map through its depth point cloud (from topic `/mynteye/points/data_raw`). I am also using the RTAB-Map to perform localization and this provides me the correct pose of the camera in the map. RTAB-Map also provides a point cloud of the FOV of the camera, so I can see the the 3D map that is captured from the FOV.
Yet, when I check the depth point cloud of the mynteye (`/mynteye/points/data_raw`), this does not overlap with the RTAB-Map, it is deviated from it. 

Even both 3 cameras placed in exact same pose, RTAB-Map is precise for both, but each camera presents a different deviation from the rtabmap point cloud.

I already have calibrated all cameras, and all error projections are below 0.2 pixels. I have used your mynteye-s-calibrator tool.

In this next image you can see the depth pointcloud of the mynteye (in white) and the rtabmpap depth point cloud (in red).
[image]

Is there any way to correct this deviation? Depth correction?

Hope this correctly expose my problem.

FaboNo, 2020-12-04:
@tiagonovo0 sorry to bother you but I would like to know if the depth point cloud was working "out of the box" or did you need to calibrate the MYNT (meaning that there is somehow no factory calibration) ?
Indeed Iooks like that point cloud is not working properly (https://github.com/slightech/MYNT-EYE-S-SDK/issues/102) and this is what we noticed in our lab as well.
Thank you for your answer

tiagonovo0, 2021-01-06:
Hi, I got this pointcloud after calibrating the Mynt. Before that, I had a few points as mentioned in the issue [#102](https://github.com/slightech/MYNT-EYE-S-SDK/issues/102).