slightech/MYNT-EYE-S-SDK #109: Segfault in DisparityProcessor

kekeblom, 2020-12-03:
I installed the sdk and roswrapper using `make -j ros`. I'm running opencv 3.4.5 as suggested by the docs.

When I `roslaunch mynt_eye_ros_wrapper mynteye.launch`, the node segfaults. When I hookup gdb and run backtrace after the crash, I get the stacktrace at the bottom. I.e. the error seems to be at `disparity_processor.cc:74`.

I can run the the `get_depth` sample, so the device and sdk is somewhat working.

I am running Ubuntu 20.04, which I realize is not officially supported, but would very much appreciate any help on this.

If there is anything I can do to help diagnose or narrow down the issue, please let me know.

Thanks! 

```
#0  0x0000000100000176 in  ()                                                                                                                                                                                      
#1  0x00007ffff4c57439 in mynteye::DisparityProcessor::DisparityProcessor(mynteye::DisparityComputingMethod, std::shared_ptr<mynteye::CameraROSMsgInfoPair>, int) (this=                                           
    0x958e40, type=mynteye::DisparityComputingMethod::BM, calib_infos=std::shared_ptr<struct mynteye::CameraROSMsgInfoPair> (empty) = {...}, proc_period=0)                                                        
    at /home/ken/crap/MYNT-EYE-S-SDK/src/mynteye/api/processor/disparity_processor.cc:74                                                                                                                           
#2  0x00007ffff4c5072d in __gnu_cxx::new_allocator<mynteye::DisparityProcessor>::construct<mynteye::DisparityProcessor, mynteye::DisparityComputingMethod, decltype(nullptr), int>(mynteye::DisparityProcessor*, my
nteye::DisparityComputingMethod&&, decltype(nullptr)&&, int&&)                                                                                                                                                     
    (this=0x7fffffffb5e0, __p=0x958e40, __args=@0x7fffffffc18f: mynteye::DisparityComputingMethod::BM, __args=<error reading variable>, __args=@0x7fffffffc17c: 0)                                                 
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/ext/new_allocator.h:147                                                                                                                    
#3  0x00007ffff4c504b7 in std::allocator_traits<std::allocator<mynteye::DisparityProcessor> >::construct<mynteye::DisparityProcessor, mynteye::DisparityComputingMethod, decltype(nullptr), int>(std::allocator<myn
teye::DisparityProcessor>&, mynteye::DisparityProcessor*, mynteye::DisparityComputingMethod&&, decltype(nullptr)&&, int&&)                                                                                         
    (__a=..., __p=0x958e40, __args=@0x7fffffffc18f: mynteye::DisparityComputingMethod::BM, __args=<error reading variable>, __args=@0x7fffffffc17c: 0)                                                             
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/alloc_traits.h:484                                                                                                                    
#4  0x00007ffff4c501f8 in std::_Sp_counted_ptr_inplace<mynteye::DisparityProcessor, std::allocator<mynteye::DisparityProcessor>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<mynteye::DisparityComputingMe
thod, decltype(nullptr), int>(std::allocator<mynteye::DisparityProcessor>, mynteye::DisparityComputingMethod&&, decltype(nullptr)&&, int&&)
    (this=0x958e30, __a=..., __args=@0x7fffffffc18f: mynteye::DisparityComputingMethod::BM, __args=<error reading variable>, __args=@0x7fffffffc17c: 0)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr_base.h:548
#5  0x00007ffff4c4ff20 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<mynteye::DisparityProcessor, std::allocator<mynteye::DisparityProcessor>, mynteye::DisparityComputingMethod, decltype(nul
lptr), int>(mynteye::DisparityProcessor*&, std::_Sp_alloc_shared_tag<std::allocator<mynteye::DisparityProcessor> >, mynteye::DisparityComputingMethod&&, decltype(nullptr)&&, int&&)
    (this=0x7fffffffc198, __p=@0x7fffffffc190: 0x0, __a=..., __args=@0x7fffffffc18f: mynteye::DisparityComputingMethod::BM, __args=<error reading variable>, __args=@0x7fffffffc17c: 0)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr_base.h:680
#6  0x00007ffff4c4fe1d in std::__shared_ptr<mynteye::DisparityProcessor, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<mynteye::DisparityProcessor>, mynteye::DisparityComputingMethod, decltype(nullptr
), int>(std::_Sp_alloc_shared_tag<std::allocator<mynteye::DisparityProcessor> >, mynteye::DisparityComputingMethod&&, decltype(nullptr)&&, int&&)
    (this=0x7fffffffc190, __tag=..., __args=@0x7fffffffc18f: mynteye::DisparityComputingMethod::BM, __args=<error reading variable>, __args=@0x7fffffffc17c: 0)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr_base.h:1344
#7  0x00007ffff4c4fd97 in std::shared_ptr<mynteye::DisparityProcessor>::shared_ptr<std::allocator<mynteye::DisparityProcessor>, mynteye::DisparityComputingMethod, decltype(nullptr), int>(std::_Sp_alloc_shared_ta
g<std::allocator<mynteye::DisparityProcessor> >, mynteye::DisparityComputingMethod&&, decltype(nullptr)&&, int&&)
    (this=0x7fffffffc190, __tag=..., __args=@0x7fffffffc18f: mynteye::DisparityComputingMethod::BM, __args=<error reading variable>, __args=@0x7fffffffc17c: 0)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr.h:359
#8  0x00007ffff4c4fcc9 in std::allocate_shared<mynteye::DisparityProcessor, std::allocator<mynteye::DisparityProcessor>, mynteye::DisparityComputingMethod, decltype(nullptr), int>(std::allocator<mynteye::Dispari
tyProcessor> const&, mynteye::DisparityComputingMethod&&, decltype(nullptr)&&, int&&)                    
    (__a=..., __args=@0x7fffffffc18f: mynteye::DisparityComputingMethod::BM, __args=<error reading variable>, __args=@0x7fffffffc17c: 0)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr.h:701
#9  0x00007ffff4c490ba in std::make_shared<mynteye::DisparityProcessor, mynteye::DisparityComputingMethod, decltype(nullptr), int>(mynteye::DisparityComputingMethod&&, decltype(nullptr)&&, int&&)
    (__args=@0x7fffffffc18f: mynteye::DisparityComputingMethod::BM, __args=<error reading variable>, __args=@0x7fffffffc17c: 0)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr.h:717
#10 0x00007ffff4c427a2 in mynteye::Synthetic::InitProcessors() (this=0xd37d40)
    at /home/ken/crap/MYNT-EYE-S-SDK/src/mynteye/api/synthetic.cc:335
```

valenbase, 2021-07-28:
how did you fix the problem?

kekeblom, 2021-07-28:
I actually never figured out the issue and just used a different sensor that we happened to have at the office. Worked great!

RaviBeagle, 2022-07-19:
I think I this same issue.

RaviBeagle, 2022-07-20:
I see that it is caused when calling an OpenCV method. 

    sgbm_matcher = cv::StereoSGBM::create(0, 16, 3);
    sgbm_matcher->setPreFilterCap(63); // SEG FAULTS at this line.

So could be be an OpenCV bug ?

How can I disabled the Disparity processor ? I dont need it.

RaviBeagle, 2022-08-01:
Found a workaround atleast:

[https://github.com/slightech/MYNT-EYE-S-SDK/issues/117](url)

nitschw, 2023-01-22:
For me it turned out to be version check related code. I rebuilt the ros wrapper with gdb debugging and found these lines. Here are my git diffs to fix it:

```
diff --git a/src/mynteyed/internal/camera_p.cc b/src/mynteyed/internal/camera_p.cc
index df52c65..70790e4 100644
--- a/src/mynteyed/internal/camera_p.cc
+++ b/src/mynteyed/internal/camera_p.cc
@@ -287,8 +287,8 @@ void CameraPrivate::EnableProcessMode(const ProcessMode& mode) {
 }
 
 void CameraPrivate::EnableProcessMode(const std::int32_t& mode) {
-  if (Version(1, 6) > descriptors_->firmware_version)
-    motions_->EnableProcessMode(mode);
+  //if (Version(1, 6) > descriptors_->firmware_version)
+  //  motions_->EnableProcessMode(mode);
 }
 
 bool CameraPrivate::IsImageInfoSupported() const {
```
```
diff --git a/wrappers/ros/src/mynteye_wrapper_d/src/mynteye_wrapper_nodelet.cc b/wrappers/ros/src/mynteye_wrapper_d/src/mynteye_wrapper_nodelet.cc
index 6e97687..99929b7 100755
--- a/wrappers/ros/src/mynteye_wrapper_d/src/mynteye_wrapper_nodelet.cc
+++ b/wrappers/ros/src/mynteye_wrapper_d/src/mynteye_wrapper_nodelet.cc
@@ -291,9 +291,12 @@ class MYNTEYEWrapperNodelet : public nodelet::Nodelet {
       NODELET_INFO_STREAM(dashes);
 
       params.dev_index = dev_index;
     }
+    /*
     spec_version = mynteye->GetDescriptors()->spec_version;
     {
       std::vector<StreamInfo> color_infos;
       std::vector<StreamInfo> depth_infos;
       mynteye->GetStreamInfos(dev_index, &color_infos, &depth_infos);
@@ -311,7 +314,7 @@ class MYNTEYEWrapperNodelet : public nodelet::Nodelet {
         NODELET_INFO_STREAM(info.index << " | " << info);
       }
       NODELET_INFO_STREAM(dashes);
-    }
+    }*/
 
     pub_mesh_ = nh.advertise<visualization_msgs::Marker>("camera_mesh", 0 );
     // where to get the mesh from
@@ -445,7 +448,7 @@ class MYNTEYEWrapperNodelet : public nodelet::Nodelet {
 
   void openDevice() {
     if (mynteye->IsOpened()) return;
-
+    std::cout << "OPENING DEVICES" << std::endl;
     // Set stream data callbacks
     std::vector<ImageType> types{
       ImageType::IMAGE_LEFT_COLOR,
@@ -749,9 +752,9 @@ class MYNTEYEWrapperNodelet : public nodelet::Nodelet {
 
   void publishAlignImu(bool imu_sub,
       bool imu_processed_sub, bool temp_sub) {
-    if (spec_version <= Version(1, 0)) {
-      timestampAlign();
-    }
+    /*if (spec_version <= Version(1, 0)) {
+      timestampAlign();
+    }*/
 
     if (imu_accel == nullptr || imu_gyro == nullptr) {
       return;
```

 This is a bandaid for sure but it got me going again. Now I run make ros and the result works just fine.