slightech/MYNT-EYE-S-SDK #110: Wrong hardware version after updating firmware

sofiane-K, 2020-12-07:
Hello, 
We are currently using your products for our robots. We have 2 cameras in which we tried to put the same version (2.7) of the S1030 firmware image. In the camera with serial number 004A042400090913, the hardware version is noted: 0.0 and we are not able to get any image with the samples. In the other S1030 camera with serial number 00000185310E00090834 everything goes very well and we are able to visualize the images. The hardware version in the second one is : Hardware version: 2.0.

I think the main problem is linked to the hardware version. I also want to add that the first time I tried to upgrade the firmware on the first camera, it didn’t work. I had to manually update Westbridge driver. firmware version was 0.0 before the firmware update, I think hardware version was 3.0 but I am not sure I remember correctly, the devices were noted as Mynteye S1040 before the update while they were S1030.

After the update, the samples to get image, depth and so on didn’t work anymore.
Could you help me with this issue ? 
Thanks a lot

randohinn, 2021-01-05:
I have the same issue...

kwads123, 2021-01-06:
I am also having the same problem. The camera came with firmware version 3.0 installed but I can't try putting this back on it because its not available online. sn: 0041051900090913

xjp99v5, 2021-02-14:
I have the same issue. Oh, no. It should not remind us to update our firmware.

urbste, 2021-05-26:
Same problem! I can not use the S1030 camera this way. Mine came with firmware 0.0 and no calibration.
Also the gyroscope data is only available in 1 degree steps?! I guess that is a firmware issue as well.