slightech/MYNT-EYE-S-SDK #111: Compile error in synthetic.cc

JohnVorwald, 2020-12-22:
System: Ubuntu 18.04
Command: make samples
Abort due to errors in line 340 of synthetic.cc

Output of "make samples"

/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc:340:26: error: no matching function for call to ‘make_shared<<expression error> >(std::shared_ptr<mynteye::CameraROSMsgInfoPair>, std::shared_ptr<int>, std::shared_ptr<int>, int)’
         DEPTH_PROC_PERIOD);
                          ^
In file included from /usr/include/c++/7/memory:81:0,
                 from /MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.h:19,
                 from /MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc:14:
/usr/include/c++/7/bits/shared_ptr.h:703:5: note: candidate: template<class _Tp, class ... _Args> std::shared_ptr<_Tp> std::make_shared(_Args&& ...)
     make_shared(_Args&&... __args)
     ^~~~~~~~~~~
/usr/include/c++/7/bits/shared_ptr.h:703:5: note:   template argument deduction/substitution failed:
/MYNT-EYE-S-SDK-master/src/mynteye/api/synthetic.cc:340:26: error: template argument 1 is invalid
         DEPTH_PROC_PERIOD);

Output of "make -n samples"

~/Downloads/MYNT-EYE-S-SDK-master$ make -n samples
text="Make uninstall"; options=""; [ -z "" ] && options="1;33"; echo -e "\033[${options}m${text}\033[0m"
sudo rm -rf /usr/local/include/mynteye/
sudo rm -rf /usr/local/lib/libmynteye.so*
sudo rm -rf /usr/local/lib/cmake/mynteye/
sudo rm -rf /usr/local/share/mynteye/
text="Make build"; options=""; [ -z "" ] && options="1;33"; echo -e "\033[${options}m${text}\033[0m"
work_dir="./_build"; build_dir=".."; [ -z ".." ] && build_dir=..; build_options=""; 		([ -e "${work_dir}" ] || mkdir -p "${work_dir}") && cd "${work_dir}" && echo -e "CD: ${work_dir}" && cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=/usr/bin/cc -DCMAKE_CXX_COMPILER=/usr/bin/c++ -DCMAKE_MAKE_PROGRAM=make ${build_options}  ${build_dir}  && make
text="Make install"; options=""; [ -z "" ] && options="1;33"; echo -e "\033[${options}m${text}\033[0m"
cd ./_build; sudo make install
text="Make samples"; options=""; [ -z "" ] && options="1;33"; echo -e "\033[${options}m${text}\033[0m"
work_dir="./samples/_build"; build_dir=""; [ -z "" ] && build_dir=..; build_options=""; 		([ -e "${work_dir}" ] || mkdir -p "${work_dir}") && cd "${work_dir}" && echo -e "CD: ${work_dir}" && cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=/usr/bin/cc -DCMAKE_CXX_COMPILER=/usr/bin/c++ -DCMAKE_MAKE_PROGRAM=make ${build_options}  ${build_dir}  && make

linhaigoo, 2021-03-02:
I meet the same problem because of opencv4. The Option.cmake file will discard camera module component when Opencv4 used (line39).However, after I commented that line,I found the code base doesn't support opencv4 indeed.

truncs, 2021-04-20:
My branch should work with opencv4 

https://github.com/truncs/MYNT-EYE-S-SDK