slightech/MYNT-EYE-S-SDK #114: Could not get img params of version 2.240, please use latest SDK

xjp99v5, 2021-02-14:
Hi,

I updated my MYNT S1030's firmware to the MYNTEYE-S1030-2.7.0.img according to your guidelines. However, the SDK cannot run anymore after I updated my firmware. 
It always shows errors as following,

~/Documents/MYNT/MYNT-EYE-S-SDK/samples/_output/bin$ ./get_img_params
I/utils.cc:35 Detecting MYNT EYE devices
F/file_channel.cc:442 Could not get img params of version 2.240, please use latest SDK.
Aborted (core dumped)

binweil, 2021-02-15:
Hi,

I solve this issue by using the  MYNTEYE-S1030-2.5.0.img with SDK 2.5.0. The SDK compatibility is a little confusing cause it was saying MYNTEYE-S1030-2.7.0.img is compatible to 2.5.0~latest, where the 2.5.0 is not compatible. Hope the they would improve the documentation in the future.

xjp99v5, 2021-02-15:
Hi,

Thank you for your suggestion. I tried to roll back img-2.5.0 last night and it works, but each time I need to choose a single device or multi devices mode, it sucks.