slightech/MYNT-EYE-S-SDK #116: Unable to run samples after firmware update

ennasros, 2021-03-02:
I recently got two Mynteye S1030 to play with. I am currently using Ubuntu 18.04 (kernel 5.4.0) I setup  [MYNT-EYE-S-SDK-2.5.0](https://github.com/slightech/MYNT-EYE-S-SDK/tree/v2.5.0) and was able to get things up and running pretty smoothly. I ran the `samples/_output/bin/camera_with_junior_device_api`, and the camera feeds displayed properly. Here is the sample of the output i got when everything was running fine

```
I/utils.cc:35 Detecting MYNT EYE devices
E/device.cc:670 Image params not found, but we need it to process the images. Please `make tools` and use `img_params_writer` to write the image params. If you update the SDK from 1.x, the `SN*.conf` is the file contains them. Besides, you could also calibrate them by yourself. Read the guide doc (https://github.com/slightech/MYNT-EYE-SDK-2-Guide) to learn more.
W/device.cc:677 Intrinsics & extrinsics not exist. Use default intrinsics.
I/device.cc:678 camera calib model: unknow, use default pinhole data
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1040, sn: 000000412D2500090913, firmware: 0.0
I/utils.cc:60 Only one MYNT EYE device, select index: 0
I/utils.cc:99 MYNT EYE requests:
I/utils.cc:102   index: 0, request: width: 752, height: 480, format: Format::YUYV, fps: 25
I/utils.cc:107 Only one stream request, select index: 0
...
```

As you can see, the firmware listed was 0.0, which game me the (then brilliant, now regrettable) idea of updating the firmware on one of the two S1030.  After going through the steps and verifying the [Firmware/SDK compatibility](https://mynt-eye-s-sdk.readthedocs.io/en/latest/src/firmware/applicable.html), I updated the firmware to 2.7 (which should be compatible with my sdk). Retesting things using the camera with the updated firmware, the same sample file doesn't work. Below is the output:

```
I/utils.cc:35 Detecting MYNT EYE devices
E/device.cc:670 Image params not found, but we need it to process the images. Please `make tools` and use `img_params_writer` to write the image params. If you update the SDK from 1.x, the `SN*.conf` is the file contains them. Besides, you could also calibrate them by yourself. Read the guide doc (https://github.com/slightech/MYNT-EYE-SDK-2-Guide) to learn more.
W/device.cc:677 Intrinsics & extrinsics not exist. Use default intrinsics.
I/device.cc:678 camera calib model: unknow, use default pinhole data
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 00000041181900090913, firmware: 2.7
I/utils.cc:60 Only one MYNT EYE device, select index: 0
I/utils.cc:99 MYNT EYE requests:
I/utils.cc:102   index: 0, request: width: 752, height: 480, format: Format::YUYV, fps: 25
I/utils.cc:102   index: 1, request: width: 376, height: 240, format: Format::YUYV, fps: 25
I/utils.cc:113 There are 2 stream requests, select index: 
0
F/streams.cc:165 Timeout waiting for key frames. Please use USB 3.0, and not in virtual machine.
Aborted (core dumped)
```

I tried rolling back to earlier SDK and Firmware versions (I tried using SDK2.4.1 and SDK2.5 with FMW2.5 without much success.

urbste, 2021-05-26:
Hi,
any success? I also just got a S1030 and it also shows Firmware 0.0. 
However I do not even get to download the firmware as the link is not working.

Ador2, 2021-08-24:
I'm having the same issue, have you solved it?

19gabriel96, 2022-03-23:
I have the same problem, I'm using sdk 2.5 and firmware 2.5, and it keeps showing me the same error, F/streams.cc:165 Timeout waiting for key frames. Please use USB 3.0, and not in virtual machine, I also changed cables and computers but nothing.

19gabriel96, 2022-03-23:
maybe someone already solved it?

gh18037581263, 2022-03-31:
I was the developer,the company was dismissed,i can save your device still,i pushed the new device firmware for MYNTEYE S1030 or S1040 in https://download.csdn.net/download/u012572552/85060955 , you can update device firmware by using MYNTEYE TOOL,for solve your problem ,Just a few points in return. GOOD LUCK!

19gabriel96, 2022-04-11:
@gh18037581263 hi how are things!
The past help on the new firmware compatible with the sdk 2.5 was very helpful and everything works fine, but now I am trying to calibrate the cameras on ubuntu 16 LTS, but it shows me the following error:
mynteye-s-calibrator -x 11 -y 7 -s 0.036
I/main.cc:68 Calibrate MYNT EYE S device with chessboard 11x7, size: 0.036m, num: 11
I/utils.cc:35 Detecting MYNT EYE devices
F/channels.cc:477 Check failed: n == file_size The firmware does not support getting device info, you could upgrade to latest
Aborted (core dumped)

I would really appreciate your help.

liu83, 2022-04-13:
@gh18037581263 
Thank you very much for providing this firmware. It also works for me now!

Andrew-rw, 2022-06-02:
@liu83 may I ask you to share the firmware image? Unfortunately I have difficulties downloading it from the resource which @gh18037581263 has mentioned:(

Thank you in advance!

EloyAldao, 2023-01-13:
@gh18037581263 We can not download the firmware outside of China, as we can not register in CSDN.
Could you upload it to an international repository like GoogleDrive or Mega please?
Thanks in advance

NguyenVanNhatHoang, 2025-04-13:
**_Hello, i have same problem as you. beside that, i also have this problem when i run ./get_stereo_rectified although i had updated firmware s1030_2.5.0 with sdk 2.5.0:_**
:~/YOLO_SLAM/MYNT-EYE-S-SDK/samples/_output/bin$ ./get_stereo_rectified 
I/utils.cc:35 Detecting MYNT EYE devices
E/device.cc:670 Image params not found, but we need it to process the images. Please make tools and use img_params_writer to write the image params. If you update the SDK from 1.x, the SN*.conf is the file contains them. Besides, you could also calibrate them by yourself. Read the guide doc (https://github.com/slightech/MYNT-EYE-SDK-2-Guide) to learn more.
W/device.cc:677 Intrinsics & extrinsics not exist. Use default intrinsics.
I/device.cc:678 camera calib model: unknow, use default pinhole data
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 0041212700090913, firmware: 2.5
I/utils.cc:60 Only one MYNT EYE device, select index: 0
I/synthetic.cc:49 camera calib model: pinhole
**Any one help me please!**