slightech/MYNT-EYE-S-SDK #117: Segmentation fault (core dumped)

shenriver, 2021-03-05:
[mynteye_1/mynteye_wrapper_node_1-2] process has died [pid 5445, exit code -11, cmd /home/bingo/Downloads/MYNT-EYE-S-SDK/wrappers/ros/devel/lib/mynt_eye_ros_wrapper/mynteye_wrapper_node __name:=mynteye_wrapper_node_1 __log:=/home/bingo/.ros/log/7aef8fae-7d83-11eb-85c6-a85e4510f61d/mynteye_1-mynteye_wrapper_node_1-2.log].
log file: /home/bingo/.ros/log/7aef8fae-7d83-11eb-85c6-a85e4510f61d/mynteye_1-mynteye_wrapper_node_1-2*.log

then i run  :rosrun mynt_eye_ros_wrapper mynteye_wrapper_node
Segmentation fault (core dumped)

who can help me?

869299603-tjdtv, 2021-06-07:
裂开，我也遇到了这个问题

valenbase, 2021-07-28:
@869299603-tjdtv 问题解决了吗？

869299603-tjdtv, 2021-07-28:
我用docker搞了个ubuntu16的环境成功解决

发自我的iPhone

------------------ Original ------------------
From: heiheihei ***@***.***&gt;
Date: Wed,Jul 28,2021 6:00 PM
To: slightech/MYNT-EYE-S-SDK ***@***.***&gt;
Cc: Davidtv ***@***.***&gt;, Mention ***@***.***&gt;
Subject: Re: [slightech/MYNT-EYE-S-SDK] Segmentation fault (core dumped) (#117)

valenbase, 2021-07-28:
@869299603-tjdtv 我用的20.04，我同事的就可以，但是我的就不行～～

869299603-tjdtv, 2021-07-28:
建议降低版本，另外建议使用docker，我可以把我调好的docker给你试试

发自我的iPhone

valenbase, 2021-07-28:
@869299603-tjdtv 我在试试吧，刚研究两天，可以在挣扎一下

869299603-tjdtv, 2021-07-28:
其实市场上有不少别的相机可以替代，mynt没有技术支持还挺难受的

发自我的iPhone

RaviBeagle, 2022-08-01:
Thanks a lot for the workaround