slightech/MYNT-EYE-S-SDK #119: Where do I get the current firmware

urbste, 2021-05-20:
Hi,

I am trying to follow the instructions to update the firmware.
However the page http://doc.myntai.com/mynteye/s/download is not loading --> Bad gateway.

Someone knows where to get the firmware?

zhouzhiwen2000, 2021-09-16:
Link：https://pan.baidu.com/s/1iU6adS2vQlOJDjf6FcPo5A 
Extraction code：cqtf 
Sorry, it's in Chinese.

zhouzhiwen2000, 2021-09-16:
https://drive.google.com/drive/folders/1eG66rmCCOMrQVYaU8FvtJOEsnEEaxYVc?usp=sharing

These are all the files I have regarding MYNT-S cameras.

BSSilvia, 2022-04-25:
> Do you have firmware of MYNTEYE_S_2.0.0_rc.img?

BSSilvia, 2022-04-25:
> https://drive.google.com/drive/folders/1eG66rmCCOMrQVYaU8FvtJOEsnEEaxYVc?usp=sharing
> 
> These are all the files I have regarding MYNT-S cameras.

Do you have firmware of MYNTEYE_S_2.0.0_rc.img?