slightech/MYNT-EYE-S-SDK #120: sdk2.5

yuban002, 2021-05-29:
/usr/include/c++/9/bits/shared_ptr.h:714:5: note:   template argument deduction/substitution failed:
/home/deepin/SLAM/MYNT-EYE-S-SDK-2.5.0/src/mynteye/api/synthetic.cc:331:27: error: template argument 1 is invalid
  331 |         POINTS_PROC_PERIOD);
      |                           ^
/home/deepin/SLAM/MYNT-EYE-S-SDK-2.5.0/src/mynteye/api/synthetic.cc:336:40: error: ‘DepthProcessor’ was not declared in this scope; did you mean ‘depth_processor’?
  336 |     depth_processor = std::make_shared<DepthProcessor>(
      |                                        ^~~~~~~~~~~~~~
      |                                        depth_processor
/home/deepin/SLAM/MYNT-EYE-S-SDK-2.5.0/src/mynteye/api/synthetic.cc:340:26: error: no matching function for call to ‘make_shared<<expression error> >(std::shared_ptr<mynteye::CameraROSMsgInfoPair>, std::shared_ptr<int>, std::shared_ptr<int>, int)’
  340 |         DEPTH_PROC_PERIOD);

yuban002, 2021-05-29:
****
[image]

yuban002, 2021-05-29:
请问如何解决

869299603-tjdtv, 2021-06-07:
规定好OPENCV 的版本，最好是3.4.2，然后查一下手册看看SDK版本和设备版本是不是匹配的。