slightech/MYNT-EYE-S-SDK #129: Segmentation fault when running sample program

RaviBeagle, 2022-07-15:
Hi all,

I am trying to run a sample after building the SDK. But I get this error:

./camera_with_senior_api 
I/utils.cc:35 Detecting MYNT EYE devices
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 00000281283A00090807, firmware: 2.7
I/utils.cc:60 Only one MYNT EYE device, select index: 0
E/synthetic.cc:367 Unknow calib model type in device: CalibrationModel::KANNALA_BRANDT
2.5.0I/utils.cc:99 MYNT EYE requests:
I/utils.cc:102   index: 0, request: width: 752, height: 480, format: Format::YUYV, fps: 25
I/utils.cc:102   index: 1, request: width: 376, height: 240, format: Format::YUYV, fps: 25
I/utils.cc:113 There are 2 stream requests, select index: 
**0**
E/synthetic.cc:139 ERROR: no suited processor for stream Stream::LEFT_RECTIFIED
Segmentation fault (core dumped)

./camera_with_senior_api 
I/utils.cc:35 Detecting MYNT EYE devices
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 00000281283A00090807, firmware: 2.7
I/utils.cc:60 Only one MYNT EYE device, select index: 0
E/synthetic.cc:367 Unknow calib model type in device: CalibrationModel::KANNALA_BRANDT
2.5.0I/utils.cc:99 MYNT EYE requests:
I/utils.cc:102   index: 0, request: width: 752, height: 480, format: Format::YUYV, fps: 25
I/utils.cc:102   index: 1, request: width: 376, height: 240, format: Format::YUYV, fps: 25
I/utils.cc:113 There are 2 stream requests, select index: 
**1**
E/synthetic.cc:139 ERROR: no suited processor for stream Stream::LEFT_RECTIFIED
Segmentation fault (core dumped)

My configuration on an Ubuntu 20.04 and OpenCV 3.4.8 is as follows:

CD: ./_build
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- HOST_ARCH: x86_64
-- Found OpenCV: /usr/local (found version "3.4.8") 
-- 
-- Platform:
--   HOST_OS: Linux
--   HOST_NAME: Ubuntu
--   HOST_ARCH: x86_64
--   HOST_COMPILER: GNU
--     COMPILER_VERSION: 9.4.0
--     COMPILER_VERSION_MAJOR: 9
--     COMPILER_VERSION_MINOR: 4
--     COMPILER_VERSION_PATCH: 0
--     COMPILER_VERSION_TWEAK: 
--   OpenCV_VERSION: 3.4.8
--     OpenCV_VERSION_MAJOR: 3
--     OpenCV_VERSION_MINOR: 4
--     OpenCV_VERSION_PATCH: 8
--     OpenCV_VERSION_TWEAK: 0
--     OpenCV_VERSION_STATUS: 
--     OpenCV_WITH_WORLD: 
--   MYNTEYE_VERSION: 2.5.0
--     MYNTEYE_VERSION_MAJOR: 2
--     MYNTEYE_VERSION_MINOR: 5
--     MYNTEYE_VERSION_PATCH: 0
--     MYNTEYE_VERSION_TWEAK: 
-- 
-- Options:
--   WITH_API: ON
--     OpenCV: YES
--     OpenCV_VERSION: 3.4.8
--     OpenCV_WORLD: NO
--   WITH_DEVICE_INFO_REQUIRED: ON
--   WITH_CAM_MODELS: ON
--   WITH_BOOST: ON
--     Boost: YES
--     Boost_VERSION: 1.71.0
--   WITH_GLOG: OFF
-- 
-- Features:
--   Filesystem: boost
-- 
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using flag -std=c++11
-- C_FLAGS: -Wall -Wextra -march=native
-- CXX_FLAGS: -Wall -Wextra -std=c++11 -march=native
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: filesystem thread 
-- Namespace: mynteye
-- RPATH: /usr/local/lib
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sxv1kor/MYNT-EYE-S-SDK/_build

RaviBeagle, 2022-07-15:
And the same error when running the ROS Driver as well:

 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /mynteye/
    mynteye_wrapper_node (mynt_eye_ros_wrapper/mynteye_wrapper_node)

ROS_MASTER_URI=http://localhost:11311

process[mynteye/mynteye_wrapper_node-1]: started with pid [186702]
[ INFO] [1657883692.633234578]: Initializing nodelet with 16 worker threads.
[ INFO] [1657883692.671538608]: Detecting MYNT EYE devices
[ INFO] [1657883692.903884768]: MYNT EYE devices:
[ INFO] [1657883692.903908601]:   index: 0, name: MYNT-EYE-S1030, serial number: 00000281283A00090807
[ INFO] [1657883692.904378073]: Only one MYNT EYE device, select index: 0
E/synthetic.cc:367 Unknow calib model type in device: CalibrationModel::KANNALA_BRANDT
E/synthetic.cc:139 ERROR: no suited processor for stream Stream::LEFT_RECTIFIED
[mynteye/mynteye_wrapper_node-1] process has died [pid 186702, exit code -11, cmd /home/sxv1kor/MYNT-EYE-S-SDK/wrappers/ros/devel/lib/mynt_eye_ros_wrapper/mynteye_wrapper_node __name:=mynteye_wrapper_node __log:=/home/sxv1kor/.ros/log/bc6a302a-03f5-11ed-8812-39d5d0376a60/mynteye-mynteye_wrapper_node-1.log].
log file: /home/sxv1kor/.ros/log/bc6a302a-03f5-11ed-8812-39d5d0376a60/mynteye-mynteye_wrapper_node-1*.log
[mynteye/mynteye_wrapper_node-1] restarting process
process[mynteye/mynteye_wrapper_node-1]: started with pid [186880]

RaviBeagle, 2022-07-15:
./get_device_info 
I/utils.cc:35 Detecting MYNT EYE devices
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 00000281283A00090807, firmware: 2.7
I/utils.cc:60 Only one MYNT EYE device, select index: 0
E/synthetic.cc:367 Unknow calib model type in device: CalibrationModel::KANNALA_BRANDT
2.5.0I/get_device_info.cc:23 Device name: MYNT-EYE-S1030
I/get_device_info.cc:24 Serial number: 00000281283A00090807
I/get_device_info.cc:25 Firmware version: 2.7
I/get_device_info.cc:26 Hardware version: 2.0
I/get_device_info.cc:27 Spec version: 1.3
I/get_device_info.cc:28 Lens type: 0000
I/get_device_info.cc:29 IMU type: 0000
I/get_device_info.cc:30 Nominal baseline: 0
I/get_device_info.cc:31 Auxiliary chip version: 0.0
I/get_device_info.cc:33 Isp version: 0

WangJitaoNAV, 2024-07-23:
> Found the reason. Even though WITH_CAM_MODELS: ON the
> 
> #define WITH_CAM_MODELS /* #undef WITH_BM_SOBEL_FILTER */
> was undef. Thats why it was not working..

Hi there, I met the same problem and wonder which head file you add this define? thanks!