slightech/MYNT-EYE-S-SDK #131: samples sdk doing fine but ROS Wrapper process die

flashYB426, 2023-02-25:
* /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    rviz (rviz/rviz)
  /mynteye/
    mynteye_wrapper_node (mynt_eye_ros_wrapper/mynteye_wrapper_node)

auto-starting new master
process[master]: started with pid [24071]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bc7ef90e-b513-11ed-b951-68545a92f16c
process[rosout-1]: started with pid [24097]
started core service [/rosout]
process[mynteye/mynteye_wrapper_node-2]: started with pid [24104]
process[rviz-3]: started with pid [24105]
[mynteye/mynteye_wrapper_node-2] process has died [pid 24104, exit code -11, cmd /home/yuan/MYNT-EYE-S-SDK/wrappers/ros/devel/lib/mynt_eye_ros_wrapper/mynteye_wrapper_node __name:=mynteye_wrapper_node __log:=/home/yuan/.ros/log/bc7ef90e-b513-11ed-b951-68545a92f16c/mynteye-mynteye_wrapper_node-2.log].
log file: /home/yuan/.ros/log/bc7ef90e-b513-11ed-b951-68545a92f16c/mynteye-mynteye_wrapper_node-2*.log

titanior, 2023-05-08:
i've met the same problem

Rosa712, 2024-04-01:
i've met the same problem
[mynteye/mynteye_wrapper_node-1] process has died [pid 76883, exit code -11, cmd /home/wuyiming/MYNT-EYE-S-SDK/wrappers/ros/devel/lib/mynt_eye_ros_wrapper/mynteye_wrapper_node __name:=mynteye_wrapper_node __log:=/home/wuyiming/.ros/log/c29d4eb8-efd6-11ee-badf-fbc023fbeb30/mynteye-mynteye_wrapper_node-1.log].
log file: /home/wuyiming/.ros/log/c29d4eb8-efd6-11ee-badf-fbc023fbeb30/mynteye-mynteye_wrapper_node-1*.log