slightech/MYNT-EYE-S-SDK #132: Timeout waiting for key frames, please use USB 3.0, and not in virtual machine. And make ros failed.

Lavenderloved, 2023-12-06:
After getting the device of S1040, I found the notice that I should upgrade the firmware. And then, I did it use theMYNT TOOL BOX. But, I found that the device cannot be used anymore after I upgraded the firmware to S1030-2.5.0, with the SDK-2.4.1 in the device. after installing the SDK (make init, make install, make samples) and input the command `./samples/_output/bin/device/camera_d` in the terminal then: 

The terminal shew issues following:

```shell
I/utils.cc:35 Detecting MYNT EYE devices
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 004A242200090913, firmware: 2.5
I/utils.cc:60 Only one MYNT EYE device, select index: 0
I/utils.cc:93 MYNT EYE requests:
I/utils.cc:96   index: 0, request: width: 752, height: 480, format: Format::YUYV, fps: 60
I/utils.cc:96   index: 1, request: width: 376, height: 240, format: Format::YUYV, fps: 60
I/utils.cc:107 There are 2 stream requests, select index: 
0
F/streams.cc:94 Timeout waiting for key frames. Please use USB 3.0, and not in virtual machine.
已放弃 (核心已转储)
``` 

Maybe, this problem is the same as the  issue #116 

and then I chose to put the problem away and to run `make ros`. It failed again with showing the following:

```shell
Make install
[  9%] Built target camera_models
[100%] Built target mynteye
Install the project..
-- Install configuration: "Release"
-- Installing: /usr/local/include/mynteye/mynteye.h
-- Installing: /usr/local/include/mynteye/global.h
-- Installing: /usr/local/include/mynteye/logger.h
-- Installing: /usr/local/include/mynteye/types.h
-- Installing: /usr/local/include/mynteye/device/callbacks.h
-- Installing: /usr/local/include/mynteye/device/context.h
-- Installing: /usr/local/include/mynteye/device/device.h
-- Installing: /usr/local/include/mynteye/device/types.h
-- Installing: /usr/local/include/mynteye/device/utils.h
-- Installing: /usr/local/include/mynteye/util/files.h
-- Installing: /usr/local/include/mynteye/util/strings.h
-- Installing: /usr/local/include/mynteye/util/times.h
-- Installing: /usr/local/include/mynteye/api/api.h
-- Installing: /usr/local/include/mynteye/api/plugin.h
-- Installing: /usr/local/include/mynteye/api/object.h
-- Installing: /usr/local/include/mynteye/miniglog.h
-- Installing: /usr/local/include/mynteye/callbacks.h
-- Installing: /usr/local/include/mynteye/context.h
-- Installing: /usr/local/include/mynteye/device.h
-- Installing: /usr/local/include/mynteye/files.h
-- Installing: /usr/local/include/mynteye/glog_init.h
-- Installing: /usr/local/include/mynteye/strings.h
-- Installing: /usr/local/include/mynteye/times.h
-- Installing: /usr/local/include/mynteye/utils.h
-- Installing: /usr/local/include/mynteye/api.h
-- Installing: /usr/local/include/mynteye/plugin.h
-- Installing: /usr/local/include/mynteye/object.h
-- Installing: /usr/local/lib/libmynteye.so.2.4.1
-- Installing: /usr/local/lib/libmynteye.so.2
-- Installing: /usr/local/lib/libmynteye.so
-- Set runtime path of "/usr/local/lib/libmynteye.so.2.4.1" to "/usr/lib/x86_64-linux-gnu"
-- Installing: /usr/local/lib/cmake/mynteye/mynteye-config.cmake
-- Installing: /usr/local/lib/cmake/mynteye/mynteye-config-version.cmake
-- Installing: /usr/local/lib/cmake/mynteye/mynteye-targets.cmake
-- Installing: /usr/local/lib/cmake/mynteye/mynteye-targets-release.cmake
-- Installing: /usr/local/share/mynteye/build.info
Make ros
/bin/bash: catkin_make: 未找到命令
Makefile:195: recipe for target 'ros' failed
make: *** [ros] Error 127
```

Anybody helps????
Much thanks!!!!!!!!!!

NguyenVanNhatHoang, 2025-04-13:
**_Hello, i have same problem as you. beside that, i also have this problem when i run ./get_stereo_rectified although i had updated firmware s1030_2.5.0 with sdk 2.5.0:_**
:~/YOLO_SLAM/MYNT-EYE-S-SDK/samples/_output/bin$ ./get_stereo_rectified 
I/utils.cc:35 Detecting MYNT EYE devices
E/device.cc:670 Image params not found, but we need it to process the images. Please make tools and use img_params_writer to write the image params. If you update the SDK from 1.x, the SN*.conf is the file contains them. Besides, you could also calibrate them by yourself. Read the guide doc (https://github.com/slightech/MYNT-EYE-SDK-2-Guide) to learn more.
W/device.cc:677 Intrinsics & extrinsics not exist. Use default intrinsics.
I/device.cc:678 camera calib model: unknow, use default pinhole data
I/utils.cc:48 MYNT EYE devices:
I/utils.cc:51   index: 0, name: MYNT-EYE-S1030, sn: 0041212700090913, firmware: 2.5
I/utils.cc:60 Only one MYNT EYE device, select index: 0
I/synthetic.cc:49 camera calib model: pinhole
**Any one help me please!**

NguyenVanNhatHoang, 2025-04-13:
hello, how can you solve F/streams.cc:94 Timeout waiting for key frames. Please use USB 3.0, and not in virtual machine. ?